OrcaSlicer/OrcaSlicer #7174: carnt reat text in dark mode

moth4017, 2024-10-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

i can not use dark mode as the text in the menus is white on a white back ground, using vere 2.2 rc1 
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

in dark mode the back ground in the drop down menus needs to be dark

Noisyfox, 2024-10-21:
What OS are you using?

moth4017, 2024-10-21:
windows 10pro

Ergonomicmike, 2024-10-21:
I use dark mode and can see the text. Win10. Currently using 2.2.0-rc. But it's never been a problem on my installation.

Perhaps a graphics driver (or setting?) problem for the OP?

moth4017, 2024-10-21:
> I use dark mode and can see the text. Win10. Currently using 2.2.0-rc. But it's never been a problem on my installation.
> 
> Perhaps a graphics driver (or setting?) problem for the OP?

hi c an i see a screen  shot  of yours ,
 of what i showed above

Ergonomicmike, 2024-10-22:
[image]

moth4017, 2024-10-22:
not sure why mine is white

Ergonomicmike, 2024-10-22:
I dunno either. There are many variables in Windows that can affect this. Accessibility, Themes, etc.

Maybe install Windows 10 pro in a VM and run the portable version of Orca to see what happens at your end.

moth4017, 2024-10-22:
tried on 3 different  windows pc/laptops , all the same

Ergonomicmike, 2024-10-22:
That is strange. Are they all using a mostly vanilla version of Windows? And that Microsoft Edge WebView2 Runtime that's necessary for Orca? (Which I think is related to displaying assets.) 

I can't remember now - is there a VisualBasic Runtime necessary for Orca too? Yes, vcredist2019_x64

(https://github.com/SoftFever/OrcaSlicer/releases/download/v1.0.10-sf2/vcredist2019_x64.exe)

Maybe re-download Orca? (In case there's a missing/corrupt file?) Maybe redownload the latest version of Edge Webview2? And make sure that the 2019 version of VB Runtime is installed. (I've seen cases where a newer version of VB Runtime no longer supports a call that the old one supported.)

I'm guessing that your locale us US English? (Since your screenshot was in English.) I can imagine how another locale might interfere with the display of text.

Any anti-virus that could intercept a WebView action?

Perhaps someone else will report that they also have this issue?

moth4017, 2024-10-24:
tried all the above same issue

Ergonomicmike, 2024-10-24:
Wow. Well, I don't have any more ideas about what could be causing your issue. Sorry.

So far you're the only person who's reported having this issue (AFAIK) - albeit now on 3 computers on your side.

Mouse driver? (Although I doubt it would be the same driver on your 3 different computers.)

moth4017, 2024-10-24:
just purchase a new windows 11 laptop ill try it on there

moth4017, 2024-10-24:
ok just installed on the win11 home laptop and it works ok,  not my laptop though.  i need to find whats common to all my 3 windows devices

moth4017, 2024-10-25:
the only way i can get to see the drop down menu correctly is to set the  PC to "high contrast" theme  and thats not going to be usable

moth4017, 2024-10-25:
ok i reinstall the windows default  theme  , and now its working 
[image]

Ergonomicmike, 2024-10-25:
Interesting. Glad that you found the problem.

So this issue can be closed?

closed 2025-02-01 by github-actions[bot]