OrcaSlicer/OrcaSlicer #7176: Randomly stops extruding during print

zexcster, 2024-10-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1, 2.2.0 RC

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

kingroon kp3s v3 (stock)

### How to reproduce

it's probably a duplicate of closed #7020 but 
not it's not related to klipper

I've tried to print this [thing](https://www.thingiverse.com/thing:3306294)
Sliced it on 2.1.1, 2.2.0RC, default settings for the printer, except: 

-  Organic supports, 
-  Mouse Ears as brim, 
-  0.12mm layer height 
-  height of the model was set to 70mm

Everytime it prints it stops extruding at random height, like 5-10mins or so later. All the rest operations work as normal, so it moves and heats, but extruder stops responding
Tried 5 times, no success, too many waste

2 other models I've tried worked fine (just installed OcraSlicer for the first time)

 

### Actual results

Stops extruding mid print

### Expected results

Should print to the end

### Project file & Debug log uploads

[spider_miles_2.1.1_(no_model).3mf.zip](https://github.com/user-attachments/files/17458357/spider_miles_2.1.1_.no_model.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-10-21:
I also have Kp3s 3.0 and don't encounter such problem. But I use klipper though.
I think it somehow connected to slice resolution. Try to increase twice resolution (in quality tab) and test it.

zexcster, 2024-10-23:
> If the machine stops extruding at random height for the same g-code file, it usually suggests it's an issue of the machine itself.

I have not experienced such an issue with Cura or Prusa slicers (lots of prints)

Also it was not random on the same file, I used different sliced g-codes, just tried to fix surface of the model and used 2 versions of the slicer (2.1.1 and 2.2.0 RC). 

Anyway I've finally managed to successfully print it using OrcaSlicer 2.2.0 RC for Linux, used the same project file, but the model was positioned on back. The previous attempts were on Windows versions + positioned vertically on legs.

Also I've printed the same model successfully using Cura before

zexcster, 2024-10-27:
UPDATE:
a new project file (model included) for reproducing
the Head part taken from [this](https://www.thingiverse.com/thing:6462852)
[head.3mf.zip](https://github.com/user-attachments/files/17534815/head.3mf.zip)

If it prints the head alone it stops extruding at about 4mm height

If I add to the project another object like the hood from the link, it prints both objects fine

Linux version 2.2.0 RC

closed 2025-02-05 by github-actions[bot]

zexcster, 2025-02-08:
Ok, so it was just ignored for bot to be able to close it. Nice.
I stopped using OCRA since then... returned back to Cura (no issues with it at all)

frank111074, 2025-02-12:
This is still remaining problem, also if you ignore it.
An STL, sliced by cura will be printed perfect, sliced by Orcaslicer will stop extruding after exact same height, with same file.
Sliced again, with orcaslicer, it stops extruding at a different height, but also the same point, whenever i print this file.
Sliced by cura, it will be printed, perfectly again

bcbmrr, 2025-05-22:
Hello I'm experiencing the exact same problem, i have 3 anycubic chiron with little to no mods and anything sliced with orca have a 75% chance to brick mid-print just like the others. Anything sliced with Cura works flawlessly.

nathanjingst, 2026-02-15:
Still an issue. Please re-open and not ignore