OrcaSlicer/OrcaSlicer #7181: Merge with Bambustudio 1.10.0

Eldenroot, 2024-10-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There seems to be a lot of nice QoL improvements and features :)

https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74

@Noisyfox @vovodroid @SoftFever 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Merge upstream changes

vovodroid, 2024-10-22:
Well, from GitHub point of view Orca is not fork of Bambu, so I guess there is no git merge procedure.

Noisyfox, 2024-10-22:
Due to different reasons, we no longer directly merge branches from BBS anymore, instead we will cherry-pick code changes individually by hand, for changes that fit our needs.

Eldenroot, 2024-10-22:
Guys, I understand and I wrote it wrong. Cherry picks is the way :)

I like changes:

1) paint brim ear
2) Edit filament information during printing
3) Auto calibration for A series
4) mesh accuracy for step files
5) ironing inset

Maybe there are more but... ;)

Noisyfox, 2024-10-22:
Just to mention that there are some other features that look interesting from previous versions (BBS 1.9.x etc.) that we haven't ported yet, which you could also take a look too.

Eldenroot, 2024-10-22:
Yes, good point, I will check but it is quite difficult to track these changea from past

cochcoder, 2024-10-22:
Thought I'd pop in and say that the Filament scarf seam settings options from the 1.10.0 beta would probably come in handy

Eldenroot, 2025-01-22:
Unstale

Eldenroot, 2025-01-25:
Yes, I hope that we will get all these handy features in next beta of orcaslicer

Noisyfox, 2025-01-26:
Yeah the coordinates selection and vertical support are on top of my list right now (other than critical bug fixes of cause).

Eldenroot, 2025-01-31:
I have A1 printer, would be great to get it.

Eldenroot, 2026-10-04:
MAybe we can close this one or update the list :)