OrcaSlicer/OrcaSlicer #7192: Task Sending to P1S lead to crash everytime.

EternityFOR, 2024-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

P1S

### How to reproduce

Everytime sending task to printer

### Actual results

Crash at 75%

### Expected results

No crash

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17476315/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

EternityFOR, 2024-10-23:
I found that this issue only happen if I switch to another window before the progress finished and switch to the device page.

closed 2025-01-30 by github-actions[bot]

keithwkx, 2025-01-31:
same issue here