OrcaSlicer/OrcaSlicer #7195: Parts moving upwards (Y) after saving and restarting

rkarlsba, 2024-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc (and others)

### Operating System (OS)

macOS

### OS Version

15.0.1

### Printer

Bambu X1C Combo

### How to reproduce

1. fill a plate with parts
2. save
3. quit
4. open orcaslicer and load the file
5. repeat

### Actual results

Parts, visible on a full bed, but probably other parts as well, move upwards (as in Y) over time
[image]

### Expected results

They stay in place

### Project file & Debug log uploads

[daabloe-bambu.zip](https://github.com/user-attachments/files/17479324/daabloe-bambu.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

No log file

brjohnso1, 2024-10-22:
Thank goodness I'm not the only one struggling with this problem. It happens most when there are a lot of plates, and more often with plates later in the series. For example, if you have plates 1-9, plates 7,8, and 9 are most likely to have parts that are shifted. It's not a consistent bug, but it is very common.

falo2k, 2024-10-25:
Same issue present in 2.1.1 for me.  Seems to only have shifted the final row of plate content.

rkarlsba, 2024-11-06:
This is becoming rather annoying. It's not only the bottom row, it's all of them and AFAICS they move a fraction of n*m, where n is something like 10mm and m is the row number, starting with 0

scrith, 2024-11-24:
Can confirm this is happening in WIndows as well with v2.2.0

It is very annoying for large projects.

kinghat, 2024-12-06:
samsies https://github.com/SoftFever/OrcaSlicer/issues/7195 ?

closed 2025-03-15 by github-actions[bot]