OrcaSlicer/OrcaSlicer #7197: Non-Full Bridges with low wall count (possible bug)

theboee, 2024-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2

### Printer

Creality K1 Max, Ender 3 S1 Pro

### How to reproduce

1. Slice long bridging model. (I used needitmakeit's burst bridging test)
2. Very the amount of walls between 1-4

### Actual results

Bridges over long distances appear to suffer from wall counts. Low walls equals a non-full bridge vs cranking up walls or 1 wall.

[image]

This is what one would expect to occur, but...

[image]

[image]

Oddly though, with just 1 wall this occurs

[image]

### Expected results

There should be a way to force bridges to be full always, regardless of wall counts. 

### Project file & Debug log uploads

[BRIDGE_BURST.zip](https://github.com/user-attachments/files/17482660/BRIDGE_BURST.zip)
[Process presets (2).zip](https://github.com/user-attachments/files/17482666/Process.presets.2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I attached my process to help with narrowing down things in case it's a setting I changed. Although I also tried with the standard profile.

theboee, 2025-01-26:
This is issue still needs fixed

igiannakas, 2025-02-10:
Unlikely the Pr above will correct this as it’s happening because the perimeter line width doesn’t leave enough space for a bridge to be performed. The bridge extrusion is always nozzle size wide (I think)

closed 2025-05-20 by github-actions[bot]