OrcaSlicer/OrcaSlicer #7198: Orca 2.2.0 RC MacOS ARM - Crashing and multicolor printing not functioning

akrtek, 2024-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 RC

### Operating System (OS)

macOS

### OS Version

Sonoma 14.4.1

### Additional system information

Sonoma 14.4.1
Model: MacBookPro18,4, BootROM 10151.101.3, proc 10:8:2 processors, 64 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix

### Printer

Bambu Lab P1P

### How to reproduce

Installed RC. 2x crashes within first 10m. Both when attempting to slice. MacOS crash report attached in files with logs below. 

Additionally, opened an existing 3MF made in Orca 2.1.1 that had color printing. Color printing is shown but when slicing, only populating one filament. 

- Tried erasing color painting and reapply, same result. 
- Exported model as stl. Quit orca. Opened orca. New project, imported the STL file, painted it. Hit slice, crashed. Repeated crashed again
- This is in addition to two prior crashes when attempting to slice

### Actual results

Crash. Report follows

### Expected results

slicing and move to preview

### Project file & Debug log uploads

[tiny 9-sided-twist-planter -3mf.zip](https://github.com/user-attachments/files/17483114/tiny.9-sided-twist-planter.-3mf.zip)
[Logs.zip](https://github.com/user-attachments/files/17483120/Logs.zip)
[Orca Crash MacOS log.pdf](https://github.com/user-attachments/files/17483136/Orca.Crash.MacOS.log.pdf)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Appreciate the help

Noisyfox, 2024-10-23:
Yeah there is a bug that if you use color 16 in painting then it crashes. Will take a look.

Noisyfox, 2024-10-23:
Please try PR https://github.com/SoftFever/OrcaSlicer/pull/7200 and see if that solves your issue.

PR #7200 (2024-10-27): Fix crash when extruder 16 is used in color painting
Fixes #7198

The issue can be easily reproduced on any platform by:
1. Add 16 filaments
2. Add a cube
3. Paint the cube using the 16th color
4. Slice & crash

Or simply slice the project from #7198.

[image]
closed 2024-10-27 by SoftFever