OrcaSlicer/OrcaSlicer #7207: Unnecessary insertion on Z height

vgdh, 2024-10-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.2.0-rc

### Operating System (OS)

Windows

### OS Version

win10

### Printer

any

### How to reproduce

slice cube

### Actual results

unnecessary z coordinate in G1 command

### Expected results

I think it should be set only once when it changed or it should be in every command

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fritzw, 2024-10-24:
How is this a problem?

vgdh, 2024-10-24:
> How is this a problem?

This can confuse some poorly written post scripts. (that is how I actually found out about this behavior)
And just to make internal perfectionist happy.

vgdh, 2025-01-24:
up

closed 2025-05-03 by github-actions[bot]