OrcaSlicer/OrcaSlicer #7210: Orcaslicer v2.2.0-rc segfaults on Mint 22 upon setting up the DEVICE tab

galvani4987, 2024-10-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_Ubuntu2404_V2.2.0-rc

### Operating System (OS)

Linux

### OS Version

GNU/Linux Mint 22

### Additional system information

AMD Ryzen 5600G, NVIDIA 1660 GPU.

### Printer

Sovol SV06 and Rook MK1

### How to reproduce

Backup and then delete ~/.config/OrcaSlicer dir;
Run orcaslicer and go thru the setup wizard;
Click the Connection icon next to the printer's name and setup "Hostname, IP or URL", hit OK;
Orcaslicer freezes and segfaults.
Go to ~/.config/OrcaSlicer/user/default/machine and edit the printer's .json config file by setting the "print_host" parameter to "".
Orcaslicer is working again!

### Actual results

Orcaslicer freezes and segfaults.

### Expected results

Orcaslicer should run normaly and gain access to my printer over the network inside de DEVICE tab.

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/user-attachments/files/17495521/Cube.3mf.zip)
[log.zip](https://github.com/user-attachments/files/17495523/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I've tested it with and without the plugin blob and stealthmode active/disabled. Same results.

el-quique, 2024-10-24:
Idem in LinuxMint 22.

tiwipewo, 2024-10-25:
Hi, I confirm this, Orca Slicer 2.2 RC (and maybe older versions)
DISTRIB_ID=LinuxMint DISTRIB_RELEASE=22 DISTRIB_CODENAME=wilma DISTRIB_DESCRIPTION="Linux Mint 22 Wilma"
Config : Xiaomi Laptop Kernel: 6.8.0-47-generic arch: x86_64 bits on  quad core model: Intel Core i5-8250U & GeForce MX150 - 8 Gb Ram

It seems to be a problem with the network plugin. I have to disable my eth network when switching from printers to avoid the segfault. Maybe due to a recent update to the BBL plugin, it was working right before.

broncosis, 2024-10-26:
[debug_Sat_Oct_26_11_26_02_55407.log.txt](https://github.com/user-attachments/files/17531646/debug_Sat_Oct_26_11_26_02_55407.log.txt)

here is my most recent log similar behavior  
works fine if I delete the bambu network plug in and only use bambu profiles but if I switch to any of my other profiles for klipper or duet machines it crashes

broncosis, 2024-10-26:
[debug_Sat_Oct_26_03_10_16_36610.log.txt](https://github.com/user-attachments/files/17532301/debug_Sat_Oct_26_03_10_16_36610.log.txt)

ok I was able to use my Duet profile fine but as soon as a chose to use my stealth changer which is klipper it crashed

el-quique, 2024-10-27:
Problem HOST Conection
From what I'm seeing, the problem comes from the Linuxmint side.
When I go to the connection settings and then click on explore, it doesn't find any device. Now I have to try it on Zorin OS

broncosis, 2024-10-27:
well as I did updates yesterday that broke it there was a bunch of webkit packages updated I am trying to find a txt file to copy and paste

darco415, 2024-10-27:
I just updated webkit on my LM22 just now hoping previous issues were addressed. Now on the RC, it flashes the home screen and it goes dark. Followed OP instructions and it worked, also device tab is working now???? Thxs Bro!

el-quique, 2024-10-27:
broncosis. Hello. You are using  LM Debian Edition? or LM Cinnamon

broncosis, 2024-10-27:
@el-quique  Cinnamon

broncosis, 2024-10-27:
as long as I don't have a hostname or ip in the Json file it seams to work ok

el-quique, 2024-10-28:
@broncosis @darco415 
**Half solution on LinuxMint 22 updated to today.**

In the host for the printer connection use:
ip or host:7125
7125 is the port for moonraker (control in moonraker.cfg file). You can upload gcode files to the printer, but we will not have them displayed in the device tab.
It might be a good idea to install:
sudo apt install libwebkitgtk-6.0-4 libwebkitgtk-6.0-de

broncosis, 2024-10-28:
ok I have been just exporting gcode and uploading it manually

broncosis, 2024-10-28:
updating web kit to version 6 like you suggested seems to have done the trick with out the port number 
but still testing

alex999888777, 2024-10-28:
@el-quique 
Thank you very much, your solution with the moonraker port works perfect for me. I never used the device tab anyway.

galvani4987, 2024-10-28:
@el-quique
Thanks! I've just downloaded the v2.2.0 release and i can confirm the issue is still there, but your solution works! 

I can send gcode to the printer over the network using <printer-ip or hostname>:7125 and OrcaSlicer works just fine.

CitrusIntellect, 2024-10-30:
#6908 #6520

hacknslay, 2024-10-30:
@el-quique 

Thank you very much, this have made my day!

Noisyfox, 2024-10-30:
I think that has to be a bug in webkit (or libwebkit2gtk) because I can put any valid url as the printer's host and it works perfectly well, as long as the url does not point to mainsail:
[image]
[image]

And as soon as I change the host to mainsail it crashes immediately.

https://github.com/mainsail-crew/mainsail/issues/1922

broncosis, 2024-10-30:
Mine points at fluid which also crashes 
Makes me kinda feel like the bug is in mainsail and fluid as it's the only thing that fails.  

⁣Get BlueMail for Android ​

Noisyfox, 2024-10-30:
They might share similar designs that triggered this issue.

tolistim, 2024-10-30:
Moved up to 2.2.0.4 release build and adding the Moonraker port 7125 to the address alleviates the SEGFAULT and allows the networking to function for sending files. However, this causes the Device tab to attempt to connect with Mainsail instead of Fluidd with a Mainsail info screen instead of the main Mainsail UI.

el-quique, 2024-10-31:
[image]
OrcaSlicer 2.2.0
Strange behavior. If I add "files" to the host address, everything works fine until I enter the Dashboard or G-Codes Viewer.
Host: klipper.local/files

PD: (WebKitWebProcess:37098): WARNING **: 21:37:47.424: The GStreamer FDK AAC plugin is missing, AAC playback is unlikely to work.

HakunMatat4, 2024-11-06:
Thank you @el-quique , that solved the problem.
I am running Mint Cinnamon 21.3 with 2.2.0 official:

sudo nala install libwebkitgtk-6.0-4 libjavascriptcoregtk-6.0-1 ( the first package pulled the second )
[IP]:7125 and mDNS works trident01.home.arpa:7125

nsandstrom, 2024-12-03:
Same thing on Ubuntu 24

Today the slicer just randomly decided to work.

Then I realized that I for once started it before I turned on the printer.

It still worked fine once the printer was started, could upload and stort jobs as usual, until i tried "reload" on the device screen.
Changing to port 7125 seems to work for now

fabiovix, 2024-12-04:
Same here, on Ubuntu 24. The chash happens both when trying fluidd on port 4408 or  mainsail on port 4409. Works fine if I point to Creality Print.   When running with debug level 5, i found the following log:

cat debug_Wed_Dec_04_02_29_38_202407.log.0 
[error]	2024-12-04 02:29:40.166533[Thread 0x0000725ea4fdc480]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-12-04 02:29:41.175920[Thread 0x0000725ea4fdc480]:02:29:41 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-12-04 02:29:41.401315[Thread 0x0000725ea4fdc480]:02:29:41 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-12-04 02:29:41.403917[Thread 0x0000725ea4fdc480]:02:29:41 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-12-04 02:29:41.404852[Thread 0x0000725ea4fdc480]:02:29:41 AM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-12-04 02:29:42.760406[Thread 0x0000725ea4fdc480]:calc_exclude_triangles:Unable to create exclude triangles

fabiovix, 2024-12-04:
Today, I managed to get it working:  i've done port forward (NAT) of the ports:  2222, 80, 9999, 5353, 4409, 4408 and 8080 on my firewall to access it outside my house.  Then, I configurated OrcaSlicer to the external IP and external port and, strangely enough, Orcaslicer worked with both Fluid (port 4408) and mansail (port 4409).   I hope this information can be helpful to solve the bug.

vaut, 2024-12-04:
Also crashes when connecting to Partially the functionality worked when installing the non-working prefix "device UI"
In Prusha slicer the same problem.
```
(orca-slicer:108111): GLib-GObject-WARNING **: 21:16:53.428: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:108111): Gtk-CRITICAL **: 21:16:53.428: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
ERROR: WebKit encountered an internal error. This is a WebKit bug.
./Source/WebKit/WebProcess/Network/WebLoaderStrategy.cpp(562) : internallyFailedLoadTimerFired
Segmentation fault (core dumped)
```

I think it will be useful to clarify: I have qidi q1 pro (fluidd)

isaacST08, 2024-12-07:
I'm having this issue too.
I can connect, upload files, and start prints if I set "print_host" to "<device-ip>:7125". But if I leave out the port number, set the port number to 4408, or set "print_host_webui" to "<device-ip>:4408" or "<device-ip>" then in all cases it crashes.
The fluidd UI in a regular browser works fine and as expected.

I'm running 
- Orca Slicer 2.2.0
- Nixos 24.11 release
- Hyprland 0.45.2
- Linux Kernel 6.12.1
- fluidd 1.31.1 
- Moonraker v0.9.3-3-gccfe32f2

Due to the fact that I'm running nixos, the expected version of webkitgtk is included along with orca-slicer and this was recently fixed to be the most correct version a few days ago (before this, orca wouldn't even install via package or app image so I can't determine if the fluidd issue occurred before this update). I can probably find the relevant issue on the nixos page if it would help, but I don't think it's a webkitgtk issue.

ToasterUwU, 2024-12-30:
Just like isaac, im also on NixOS (Unstable) and have the exact same issue as well.

svenakela, 2025-01-03:
I encountered this on Ubuntu 24.04 with all possible images, including the daily build. I have a Voron configured and in the file `~/.config/OrcaSlicer/user/default/machine/<my machine name and settings name>.json` there is a config:
    `"print_host": "http://voron1.home/"` 
which I changed to:
    `"print_host": "http://voron1.home:7125/"`

And that stopped the crashing. So @el-quique's fix works.

cemkod, 2025-01-05:
I'm having this problem both with flatpak and appImage install of nightly and latest versions.

Beagon, 2025-01-06:
I'm encountering this with the flatPak and Appimage on Manjaro.

svenakela, 2025-01-07:
@cemkod, @Beagon did you try the fix posted just above your comments?

cemkod, 2025-01-07:
@svenakela Yes, that's how I'm using it right now. But it was nice to have the mainsail screen. So it is a temporary clutch at best.

Beagon, 2025-01-07:
> @cemkod, @Beagon did you try the fix posted just above your comments?

Yes that workaround luckily works! I just wanted to let my voice be heard as to emphasize this as an issue :)

deployor, 2025-01-19:
Encountering problems when using fluidds port! With my k1 max still! Mainsail works, but the device tab is useless then!

jalexandre0, 2025-01-23:
Having this problem on 2.2.0 Nix Os installation. Add moonraker port seems to fix the problem

jalexandre0, 2025-01-23:
Same problem when using mainsail or fluid port.  I don't use the device tab often, and change to moonraker port fix my problem for now.  Nix OS 24.11 and Orca Slicer  2.2.0

klack, 2025-01-26:
Thank you for the work around

jarrard, 2025-01-27:
Putting machine IP in crashes Orca.  No work around works.

astrohoward, 2025-02-12:
Same issue on Linux Mint LDME 6, segfaults if IP address put in for Neptune 4+ printer, fluidd interface. Same whether normal user or sudo. 
Able to run Orcaslicer if no network address assigned to device. Can slice and save gcode. Upload to printer via Fluidd web interface and prints fine.

novaplusplus, 2025-02-16:
Same occurring on Kubuntu 24.10 out of the blue until I manually edited the machine config (after a LOT of trial and error) to use the direct port on the slicer URL. So from `https://hostname` to `http://hostname:7125`

I am not sure what changed as just a few hours ago it was working, but I presume it's something to do with a package update to webkit, which seems to be the absolute bane of slicers lately. Annoying as someone who doesn't even use the device tab and just has a browser bookmark to the web interface instead 🤷

jhstatewide, 2025-02-16:
This was driving me nuts. I can confirm pointing it at :7125 is a great workaround for now. I will say, I NEVER use the device tab and it's just a memory suck for me. I would love to be able to disable it entirely.

witc-git, 2025-02-23:
I'm having this on the 2.3.0 Flatpak Beta on Fedora 41, Ryzen 5, Nvidia 1060 with proprietary drivers.

etrigan63, 2025-02-24:
This is happening to me on my main workstation (CachyOS (Arch) + RTX A2000 (Nvidia proprietary drivers)), but not on my Framework 13 AMD laptop (CachyOS (Arch) + AMD M780 (kernel drivers)) it works perfectly fine.

rilkman, 2025-02-28:
getting the following with ubuntu 22.04 app image on pop os

`ERROR: WebKit encountered an internal error. This is a WebKit bug.
./Source/WebKit/WebProcess/Network/WebLoaderStrategy.cpp(562) : internallyFailedLoadTimerFired`

gaspatxo, 2025-02-28:
does anyone know what is the specific [bug in webkit](https://bugs.webkit.org/) causing this?

tarelda, 2025-03-07:
I can also confirm that issue persist on 2.3.0-beta2. My workstation setup is - i7-4790K + GTX960 on Ubuntu 24.04. NVIDIA pattern also present here. Putting IP causes instant crash. Also adding flag `--debug 5` doesn't increase the verbosity of logging in case of AppImage.

edco, 2025-03-11:
I'm seeing the same thing with a locally built AppImage and app, using both the 2.3.0-beta2 tag and the latest source (as of a day or two ago). My system is Ubuntu 24.10 running on Intel with an NVIDIA GPU. I'm a newcomer to gdb so I may be completely on the wrong track, but got it to generate the trace below. Interestingly the references to `enterAcceleratedCompositingMode` persist even with the environment variable `WEBKIT_DISABLE_COMPOSITING_MODE=1`.

```
(process:53299): libsoup-DEBUG: 22:59:52.152: stopping input source
(process:53299): libsoup-DEBUG: 22:59:52.152: received frame 1 with 1489 payload
(process:53299): libsoup-DEBUG: 22:59:52.152: message: delivering 1 with 1489 length
(process:53299): libsoup-DEBUG: 22:59:52.152: received frame 1 with 116 payload
(process:53299): libsoup-DEBUG: 22:59:52.152: message: delivering 1 with 116 length

Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
Downloading source file /usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp...
webkitWebViewBaseEnterAcceleratedCompositingMode () at /usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp:2938
warning: 2938	/usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp: No such file or directory
(process:53299): libsoup-DEBUG: 22:59:52.390: stopping input source
(process:53299): libsoup-DEBUG: 22:59:52.390: received frame 1 with 138 payload
(process:53299): libsoup-DEBUG: 22:59:52.390: message: delivering 1 with 138 length
#0  webkitWebViewBaseEnterAcceleratedCompositingMode () at /usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp:2938
#1  0x00007ffff23efad1 in WebKit::WebPageProxy::enterAcceleratedCompositingMode () at /usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/UIProcess/WebPageProxy.cpp:10743
#2  0x00007ffff25528b5 in WebKit::DrawingAreaProxyCoordinatedGraphics::enterAcceleratedCompositingMode () at /usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/UIProcess/CoordinatedGraphics/DrawingAreaProxyCoordinatedGraphics.cpp:237
#3  0x00007ffff203a5b5 in IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}::operator()<unsigned long, WebKit::LayerTreeContext>(unsigned long&&, WebKit::LayerTreeContext&&) const () at /usr/src/webkit2gtk-2.46.6-0ubuntu0.24.10.1/Source/WebKit/Platform/IPC/HandleMessage.h:135
#4  std::__invoke_impl<void, IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}, unsigned long, WebKit::LayerTreeContext>(std::__invoke_other, IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}&&, unsigned long&&, WebKit::LayerTreeContext&&) () at /usr/include/c++/14/bits/invoke.h:61
#5  std::__invoke<IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}, unsigned long, WebKit::LayerTreeContext>(IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}&&, unsigned long&&, WebKit::LayerTreeContext&&) () at /usr/include/c++/14/bits/invoke.h:96
#6  std::__apply_impl<IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}, std::tuple<unsigned long, WebKit::LayerTreeContext>, 0ul, 1ul>(IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&)::{lambda((auto:1&&)...)#1}&&, std::tuple<unsigned long, WebKit::La [... cut ...]
        __func__ = "g_main_loop_run"
#26 0x00007ffff7408025 in gtk_main () at ../../../gtk/gtkmain.c:1329
        loop = 0x55555c067730
#27 0x0000555559e571c5 in wxGUIEventLoop::DoRun() ()
#28 0x0000555559f840b1 in wxEventLoopBase::Run() ()
#29 0x0000555559f4de52 in wxAppConsoleBase::OnRun() ()
#30 0x0000555559fd6038 in wxEntry(int&, wchar_t**) ()
#31 0x0000555556eb6ac4 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#32 0x00005555561c26ab in Slic3r::CLI::run(int, char**) ()
#33 0x00005555560e43d8 in main ()
```

edco, 2025-03-12:
In my case at-least, this appears to relate to https://github.com/SoftFever/OrcaSlicer/pull/8373 - though since this issue predates that PR by several months it's clearly not the full story. Regardless, a fix or workaround that has proved robust for me is simply commenting out `export WEBKIT_DISABLE_DMABUF_RENDERER=1` here: https://github.com/SoftFever/OrcaSlicer/blob/79a5a9b5f85895ae24fad8b5070c64b495e1f522/src/platform/unix/BuildLinuxImage.sh.in#L60

I've just tested against the new v2.3.0-rc tag, running the app directly and in an AppImage, and it works flawlessly, with hardware accelerated views in the Prepare and Preview tabs, plus the Web UI in the Device tab.

If you're not building the app yourself you can edit the file that that code ends up in, `AppRun` (though if you're using a pre-built AppImage, editing that file may be a little involved).

Paging @klylabs - do you have any thoughts about this? It seems like your PR mostly fixed this issue, and getting rid of `WEBKIT_DISABLE_DMABUF_RENDERER=1` completes the fix (for me, at-least).

edco, 2025-03-13:
One `apt-get upgrade` later and my workaround seems to have fallen apart. Now it seems like involving Zink at all breaks most of the UI and spits a lot of `Failed to get GBM device` lines into the log. Instead I've arrived at the following command using the (unmodified) app binary runner:

`ZINK_DISABLE_OVERRIDE=1 __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./build/package/AppRun`

And this for running in a firejail:

`firejail --noprofile --env=ZINK_DISABLE_OVERRIDE=1 --env=__GLX_VENDOR_LIBRARY_NAME=mesa --env=__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json --appimage OrcaSlicer_Linux_V2.3.0-rc.AppImage`

gmoccapy, 2025-03-27:
I tried both the suggestion to update webkit and also the firejail command.
Using the port 4408 to archive the printer GUI from Creality K2 leads to crash the software.

If I use port :7125 I can see Moonraker GUI on the device tab. As soon as I change to 4408 as it should be, the software crashes.

Any advice what more I may test?

michas79de, 2025-03-27:
> I tried both the suggestion to update webkit and also the firejail command. Using the port 4408 to archive the printer GUI from Creality K2 leads to crash the software.
>
> If I use port :7125 I can [...]

Nothing. Just use port 7125. You'll be able to upload your sliced data / gcode files to the printer.
Then you can use the browser with port 4408 (or without port, which means port 80).

MWP, 2025-03-30:
I'm see the same problem with the official 2.3.0 appimage running on LinuxMint 22 (Wilma).

I removed the OrcaSlicer config directory, add new printer, add network connection (local/lan), go to device tab and Orca segfaults.
Running Orca again immediately segfaults. 
I have to remove the OrcaSlicer config directory again to get it to run.

witc-git, 2025-03-31:
@MWP Do you have physical printers setup in the connection tab? Are you using the Moonraker port? I got mine working by switching to Moonraker port as per previous comments, along with a collection of environment variables from various threads.
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`
`WEBKIT_DISABLE_DMABUF_RENDERER=1`
`GALLIUM_DRIVER=zink`
`MESA_LOADER_DRIVER_OVERRIDE=zink`
`__GLX_VENDOR_LIBRARY_NAME=mesa`
`WEBKIT_DISABLE_COMPOSITING_MODE=1`
`WEBKIT_FORCE_COMPOSITING_MODE=1`

MWP, 2025-03-31:
> MWP Do you have physical printers setup in the connection tab? Are you using the Moonraker port? I got mine working by switching to Moonraker port as per previous comments, along with a collection of environment variables from various threads.

I was using no port, or 10088 (i believe they are the same thing, Qidi Plus4).
I have now switched to the Moonraker port which does work fine without setting any of those env vars. Obviously not the idea solution though.
I guess something on the Fluidd UI kills the in-built browser?

tarelda, 2025-03-31:
I run AppImage in the terminal and 2.3.0 beta2 had this lines at the end of the output:

```
./Source/WebKit/WebProcess/Network/WebLoaderStrategy.cpp(562) : internallyFailedLoadTimerFired
ERROR: WebKit encountered an internal error. This is a WebKit bug.
```

It's the same observation as in #7287 , but in my case it was present in normal AppImage, but is not present in 2.3.0 stable. Adding WEBKIT_DISABLE_COMPOSITING_MODE=1 also shows these logs. It looks like enabling compositing only hides the underlying issue.

> `ZINK_DISABLE_OVERRIDE=1 __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./build/package/AppRun`
>

Adding this set of flags also causes crash, but with few new log lines:
```
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
```

@edco Are you running X11 or Wayland GNOME session? I am running pretty old, subsequently upgraded install so I am on X11 session. 

> I guess something on the Fluidd UI kills the in-built browser?

I tested against fluidd UI running in docker and it loaded fine. It seems that it has to do something when fluidd is a front for moonraker.
I decided to conduct small experient to verify that. I set printer address to some bogon in my printer config. Then I launched Orca through gdb with devtools enabled. Then through js console of devtools I navigated to my printer fluidd interface `location.href='http://correct_printer_address'` . Fluidd loaded seemingly without connection to Klippy. 

[image]

Then Orca crashed with SEGFAULT. Since I haven't bothered to load symbols my trace is much more obscure than @edco 's 

```
(gdb) bt
#0  0x00007ffff28b4476 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#1  0x00007ffff278f883 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#2  0x00007ffff28e33a7 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#3  0x00007ffff241b219 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#4  0x00007ffff26fe829 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#5  0x00007ffff2810652 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#6  0x00007ffff26f9235 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#7  0x00007ffff26f9bfc in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0
#8  0x00007ffff15583e5 in ?? () from /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0
#9  0x00007ffff15ead1d in ?? () from /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0
#10 0x00007ffff15ebb1d in ?? () from /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0
#11 0x00007ffff6d3248e in g_main_dispatch (context=0x55555b6ebce0) at ../../../glib/gmain.c:3344
#12 0x00007ffff6d91717 in g_main_context_dispatch_unlocked (context=0x55555b6ebce0) at ../../../glib/gmain.c:4152
#13 g_main_context_iterate_unlocked.isra.0 (context=0x55555b6ebce0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:4217
#14 0x00007ffff6d32f77 in g_main_loop_run (loop=0x55555be53ad0) at ../../../glib/gmain.c:4419
#15 0x00007ffff73feb45 in gtk_main () at ../../../gtk/gtkmain.c:1329
#16 0x0000555559b05615 in wxGUIEventLoop::DoRun() ()
#17 0x0000555559c312b1 in wxEventLoopBase::Run() ()
#18 0x0000555559bfb8df in wxAppConsoleBase::OnRun() ()
#19 0x0000555559c82fc8 in wxEntry(int&, wchar_t**) ()
#20 0x0000555556d137eb in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#21 0x00005555560d6c6b in Slic3r::CLI::run(int, char**) ()
#22 0x0000555555ff7958 in main ()
```

Then I tried again with navigating to my locally run fluidd instance `  `. 

[image]
 
Orca haven't crashed, so something causing crash comes from fluidd operation after estabilishing websocket connection. To further confirm my suspicion I installed epiphany-browser (which is primary target for WebKitGTK). And as you probably guessed it  crashed when trying to open printer UI. But catch is that I had to authenticate FIRST to cause crash. Login screen loads fine and probably in Orca would to, but I have no idea how to clear cookies/localstorage there. 

Summing it all up, I think I confirmed that issue stems from WebKitGTK bug (since epiphany-browser behaves exactly the same), but not how Orca sets up WebView. Moonraker port workarounds work because Prepare/Preview tabs use OctoPrint class that works with libcurl to do api calls so is not affected by WebKit bug.  https://github.com/SoftFever/OrcaSlicer/blob/f6dc9c1ab48042c7499da09c274631f5a445bd78/src/slic3r/Utils/OctoPrint.hpp#L18

Bottomline is that I believe this issue should not prevent slicer from starting. It is not neccessary for GUI to preload Device tab during initialization.

Noisyfox, 2025-04-01:
> Bottomline is that I believe this issue should not prevent slicer from starting. It is not neccessary for GUI to preload Device tab during initialization.

Yes, that's what I'm doing in PR https://github.com/SoftFever/OrcaSlicer/pull/9031

edco, 2025-04-02:
> [@edco](https://github.com/edco) Are you running X11 or Wayland GNOME session?

Wayland.

tarelda, 2025-04-02:
> > [@edco](https://github.com/edco) Are you running X11 or Wayland GNOME session?
> 
> Wayland.

So only thing we have in common is Nvidia GPU with proprietary driver.

[41 more replies]