OrcaSlicer/OrcaSlicer #7212: Organic support floating in the air

knower-to-be, 2024-10-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Biqu Hurakan

### How to reproduce

1. Slice the model with some basic settings:
 - Layer height: 0.2 mm
 - Line width: 0.4 mm
 - Organic support (auto)
 - Threshold angle: 30 degrees
 - 0 interface layers (support floats even with interface layer, but that is how I sliced).

2. Observe that some parts of the support appear to be floating in the air

### Actual results

Parts of the support appear to be floating in the air.

[image]

### Expected results

The support should be attached to the buildplate rather than floating in the air. The image below was obtained when I set parameters:
 - Tree support branching angle equal to 60 degrees (maximum value) 
 - Preferred branch angle equal to 50 degrees

[image]

Here is the gif comparing two prints layer by layer. The right one used the normal branch angle settings, while the left one uses the branch angle settings listed above:

[image]

Other parameters have a minor effect and do not eliminate floating supports completely. In particular reducing branch density from 5% to 0% helped a bit, but not as much as increasing the branching angle.

### Project file & Debug log uploads

[orca_issue.zip](https://github.com/user-attachments/files/17497424/orca_issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I found a similar issue #4158 from almost a year ago, however it appears that it has not been resolved yet.

The model has a tiny cube at the bottom to allow the model to be printed away from the buildplate. I am using the model to test different support settings, because I want to improve the supported surface quality.

gshpychka, 2024-10-25:
> I found a similar issue https://github.com/SoftFever/OrcaSlicer/issues/4158 from almost a year ago, however it appears that it has not been resolved yet.

yup - this issue seems to be a duplicate

JaredHale, 2024-12-12:
Something else you can do if needed is add a cylinder object and duplicate it around to break up the empty space. Sometimes this is needed under sharp overhang object parts as well(I print a lot of miniatures and the branch angle approach isn't always enough). Also with organic supports to get a top interface you have to paint the support on, otherwise it won't create interface.

closed 2025-03-21 by github-actions[bot]