OrcaSlicer/OrcaSlicer #7213: Request:  "Tall Print" mode - speed throttling as model gets taller

fooshards, 2024-10-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

As a print is completed, the model's center of gravity moves further and further up the Z of the print as more layers are added.  For a bedslinger-type printer, the mass of that object is being pushed around and can lead to a number of problems if too much force is applied during those actions.  Usually it results in some combination of wobble/flex in the print/plate/gantry, causing the work surface to become non planar, the nozzle scraping into the part, the part falling off, and much gnashing of teeth and cursing of words because it literally happens at the very final parts of your print.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Slowing down the print helps, but doing so slows down the bottom of the print as well, where it is already well-adhered and with a low center of gravity.  I can see at least three solutions:

1) Do complex math and calculate the model's center of mass, every layer, and when it crosses some threshold of force being applied to it by the speed/accel of (usually) Y movements, slow the sliced movements down to get closer to the threshhold.   Maybe two modes, one that tries to keep a constant extrusion flow rate on the layer, and one that only respects the instant forces.

2) Do caveman math and say that above a certain Z, begin applying a linear reduction of speed for every next layer.  It could be configurable via "First height-slowed layer", "Speed reduction at max height", etc.    Ex: Layer 300 at 100% speed, 301 at 99.8%, 302 at 99.6%, ... 1000 at 20%.

3) Do the above, but feed it into the Accel / Jerk limits for each layer, and then rely on that influencing the sliced movements.

### Describe alternatives you've considered

1) Set an alarm for midway thru the print, then manually reduce the print's speed live on the printer interface.  Doing it like this usually results in a visible line where the extrusion flow changes dramatically, though.
2) Slow down the whole print to a limit that the top of the print can endure, and eat the extra time.

### Additional context

Bedslingers doing bedslinger things.  For models with a lot of meat on the bottom, slowing down the whole print just to preserve the little spires on top makes me a sad panda.

Thanks for reading.

t-moennich, 2024-11-21:
I own multiple Neptune 4 Max and suffer from this issue a lot while trying to print large objects with 3 - 6 kg of Filament (incl. Supports).
I would love to have a more general solution to it 🙌🏼

VigilantGardener, 2024-11-22:
There's an open project that's a plugin for Cura that does this.

https://github.com/VMaxx/SlowZ

netweaver1970, 2025-02-26:
I'm interested in this as well, as my tall, lampshade style, vase mode prints are not that nice anymore 'on top', with increasing Z.

sl0m0ZA, 2025-04-11:
I would also like this as a feature for my bedslinger running klipper and using Cura

AEhere, 2025-10-31:
I would also appreciate this feature to stop messing with modifiers and manual gcode adjustments for tall prints.

Alan-Cha, 2025-12-12:
I also want this feature. I'm doing a vase mode print on a tall print (almost 1 meter tall). I would like the print speed to slow down near the top as it begins to wobble more. I tried using a height range modifier with vase mode but OrcaSlicer (2.3.2-dev) throws an error:

Spiral (vase) mode does not work when an object contains more than one material.