OrcaSlicer/OrcaSlicer #7214: Not launching on MX-23

giubu, 2024-10-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

MX-23

### Printer

Custom

### How to reproduce

I tried different Appimages, none seem to work. Some of them show the initial launch window, but they all crash before the slicer starts.

### Actual results

Application does not launch

### Expected results

expect it to work

### Project file & Debug log uploads

'/home/boombox/Downloads/OrcaSlicer_Linux_Ubuntu2004_V2.2.0-beta2.AppImage' 
[2024-10-24 10:19:31.258001] [0x00007fa5b6723440] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.145: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.145: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.147: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.147: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.148: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.148: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.149: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.149: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.151: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.151: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.152: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.152: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.154: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.154: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.155: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.155: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.156: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.156: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): GLib-GObject-WARNING **: 10:19:35.157: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.157: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.428: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.432: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.433: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.442: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.443: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.443: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.573: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.573: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.586: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.586: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.591: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.591: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.599: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.599: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.607: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.607: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.616: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.616: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.625: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.625: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.632: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:69375): Gtk-CRITICAL **: 10:19:35.632: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dmeziere, 2024-10-24:
Same for me. The same version worked well a few days ago, but today, core dump, any version (2.1.1, 2.2.0 beta).

dariusve, 2024-10-25:
In my case I had the Wayland enabled, I disable it and now it works!

dariusve, 2024-10-25:
Solved!

I had Wayland enabled but I was on X11, after switching to Wayland it works again

dariusve, 2024-10-25:
I tried this solution, it worked for few moments, now isn't working

dmeziere, 2024-10-26:
Nightly build of [OrcaSlicer_Linux_V2.2.0-rc.AppImage](https://github.com/SoftFever/OrcaSlicer/releases/download/nightly-builds/OrcaSlicer_Linux_V2.2.0-rc.AppImage)
136 MB
20 hours ago

Seems to fix this bug for me (Ubuntu Mate 20.04.5 LTS)

dariusve, 2024-10-26:
I made the app (2.1.1) keep running on Wayland, this is how it looks, did you see what's missing?
[image]

Yep, the plate is not rendered.

To make it work I have to delete the OrcaSlicer configuration folder.

dmeziere, 2024-10-26:
I am autist, so i am frequently wrong about people's intentions, but are you reproaching it to me ? If it's the case, know that i am only a user like you, i am not part of the project.

dariusve, 2024-10-26:
I'm not reproaching at you, sorry if you feel in that way. I was also trying to solve an issue I have with the X-Axis and the level of the heated bed.

I'm just sharing that OrcaSlicer is not working at all in Wayland for me.

I discovered a few things:

- The app, both v2.1.1 and the latest nightly version of the v2.2-rc, work if you remove the configuration files.
- As soon I add the address of my Klipper RPi the app crash until I remove the address, this issue occurs on Wayland and X11 for both apps
- The plate is only shown on X11

I'm think the issue, at least on my side is caused by COSMIC

Right now I'm using PrusaSlicer which works with no issues at the moment.

Sorry, again, if you feel attacked with my previous post, it wasn't my intention.

dmeziere, 2024-10-27:
Here we are again, what worked yesterday don't work no more today ...

dariusve, 2024-10-28:
Try X11

BlackShadeOSS, 2024-10-28:
I me too, I'm running Arch Linux with Hyrpland and there is no plate

Dammic, 2024-10-30:
I have the same issue with the missing plate. Fedora 40, nvidia 560 drivers.
What worked for me is this comment: https://github.com/SoftFever/OrcaSlicer/issues/5906#issuecomment-2297122044

dariusve, 2024-10-30:
That did the trick! 
Thanks!

BlackShadeOSS, 2024-11-02:
That fixed it! 
Thanks so much

Robbvy, 2025-02-09:
I tried to install Orca Slicer Windows via Wine. It works, I can import objects and see the 3D. But there is a problem with the machine selection. So it is not fully functional.
Does Orca Slicer work natively on Mx Linux for you?

Firnthon, 2025-02-09:
Gear Lever + OrcaSlicer AppImage

closed 2025-05-18 by github-actions[bot]