OrcaSlicer/OrcaSlicer #7218: The seam position set to Back does not generate a correct straight seam exactly at the back of the model.

Koboldzik, 2024-10-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The back option in the Seam position section does not generate vertical straight seams, only slightly diagonal ones and not exactly at the back of the model. In the case of some models that include seam relieve features (such as parts for the Ender 5 to Mercury One conversion), it is impossible to print them correctly. In the case of such small elements, drawing seams manually is terribly irritating and sometimes even impossible.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

If I choose a seam position set to back, I want it to be in that position. Straight, not diagonal. I'm mainly talking about holes, cylinder-like shapes, and oval walls.

### Additional context

[image]
[image]
[image]

Splarkszter, 2024-12-16:
Isn't that a result of scarf joint?

yeahdino, 2025-04-04:
Seam alignment or painting does not appear to work with a partial scarf, although it does with the "around entire wall" option selected. This bug isn't present in BambuStudio.

[image]

jpmcgillivray005, 2026-10-13:
I see this hasn't gotten much movement and I don't know why. It is still very much a problem. Back does not reliably place the seams on the back. Back aligned also doesn't. That one might actually be worse because it usually goes to the sides for some reason. Looking at Bambu studio vs orca, the seams are all over the place and way more obnoxious. Even seam painting doesn't block the seams from being put on the front.

Orcaslicer:
Aligned back: 
[image]
[image]

Back:
[image]
[image]

Bambu studio:
Back:
[image]
[image]

I did replicate all my settings from bambu studio for scarfing and it didn't change anything. I also have the entire front of the model marked red blocking seams and it still puts them there.