OrcaSlicer/OrcaSlicer #7221: Feature Request: Set Printer, Filament and Process Profiles per plate in a project.

richard-scott, 2024-10-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would like to be able to automatically set different profile settings per plate for Printer, Filament and Process profiles.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I regularly print the same model on a BBL X1C and on a BBL P1S. It is a pain to reconfigure the project each time I change printers as I have to change Printer, Filament and Process profiles. Being lazy, it would be great if each plate remembered the last settings used when it was printed.

When you select a different plate, the settings on the left half of the screen should automatically update to reflect the settings used on that plate.

### Describe alternatives you've considered

Saving multiple projects is the only way around this currently.

Aurenugeth, 2024-11-04:
I also would like to see this capability but for slightly different reasons.  

I regularly assemble print plans of 6-15 plates in a project that I work through over days of printing.  Sometimes adding and or altering future plates as I go.  Many plates have varying filament, printer or process settings that I have to change at print time instead of being able to set them up in advance.  Keeping these as separate files is difficult.  Having all the plates in the same view when assembling a print plan is much more productive.

In the interim, it would also be useful if individual plates had their own freeform notes field.

patrezp, 2024-11-24:
I would like to see this feature as well. Current multi filament setup is cumbersome, when you are using different filament as first layer. I am also using variable purge line from TeachingTech https://www.printables.com/model/1035759-adaptive-purge-for-any-3d-printer-using-slicer-var and the purge line simply does the color whichever is first in the filament section.
[image]

closed 2025-03-02 by github-actions[bot]