OrcaSlicer/OrcaSlicer #7223: Wrong Material Names in AMS

Bupher, 2024-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Printer

P1S

### How to reproduce

Change set material in AMS to PLA Basic.

https://youtu.be/H4ubGjvqOJg

### Actual results

Material sets to ABS, syncing slicer with extruder shows PLA in the slicer, but won't print as the AMS reports ABS.

### Expected results

Material sets to PLA basic, syncs PLA basic, and prints.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17515756/log.zip)
[projectForBug.zip](https://github.com/user-attachments/files/17515759/projectForBug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

edgarbs1998, 2024-11-02:
Same issue here on an A1. However, for me it does not show ABS but ASA and it is stuck on it, I just can't change to Generic PLA. If I change on the machine Bambu PLA Basic it detects PLA on Orca. On Bambu Studio it is fine.

dartrax, 2024-11-16:
I have the same problem. There is a good video showing the problem in #7432.

- I have a K Factor chosen, but still shows as ABS
- On the printer, it is correctly labelled PLA
- It prevents me from printing (can't start the Print as the ABS Slot in the AMS can't be selected) 
- Refreshing the Printer List in the Print Dialog does not help
- The workaround is to send the file to the printer and start printing at the printer

dartrax, 2025-02-15:
unstale

SoftFever, 2025-03-29:
Do you still experience this issue in 2.3.0 version?

dartrax, 2025-03-29:
Yes!

SoftFever, 2025-03-30:
> Yes!

Thank you for the updates. I can't reproduce it. 

Meanwhile, let's move our discussion to another slide. #4690
I'd like to collect more information to help me diagnose.

Smerks101, 2025-09-27:
still happening with latest beta version