OrcaSlicer/OrcaSlicer #7227: Strange top layers

jezdinskyjiri, 2024-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

X1C

### How to reproduce

[image]
[image]
It does strange move for no reason, this does not happen in BambuStudio.
Model is STEP from SolidWorks, designed for 0.2 layer height.

### Actual results

Expect to follow monotonic pattern

### Expected results

Monotonic patttern but with random artefact

### Project file & Debug log uploads

[F03A.zip](https://github.com/user-attachments/files/17521845/F03A.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-10-25:
Reduce one wall top surface threshold. Also upgrade to 2.2 RC.

vgdh, 2024-10-25:
I've sliced your project and it ok
[image]

jezdinskyjiri, 2024-10-29:
I can confirm that stable 2.2 does not have this issue. Thanks

dbuezas, 2024-11-05:
I still have the issue, even after setting the one wall threshold to 0 in v2.0.0
Right below text or other small features
[image]

igiannakas, 2024-11-05:
If you notice the preview it’s actually printing over them, not around them. This won’t cause an issue as the feature is too small to impact the nozzle.

dbuezas, 2024-11-05:
The issue I have with this is that the flat surface is not that clean because it has stops and starts.
[image]
This could be printed totally flat and results in better quality. I worked around this by disabling one wall on top layer on a modifier part that only affects that surface.

closed 2025-02-11 by github-actions[bot]