OrcaSlicer/OrcaSlicer #7229: Objects on plates get moved when opening project

julie777, 2024-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

A1mini

### How to reproduce

1. fill many plates with objects
2. close Orca Slicer
3. open Orca Slicer

### Actual results

Objects get moved so that they are no longer centered on the plates.

[image]

### Expected results

Objects should stay where I put them.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

N/A

kinghat, 2024-12-06:
i have this as well after reopening my project files:

[image]

v2.2.0 appimage

kinghat, 2024-12-06:
samsies https://github.com/SoftFever/OrcaSlicer/issues/7195 ?

closed 2025-03-15 by github-actions[bot]