OrcaSlicer/OrcaSlicer #7236: Segfault on startup

alex999888777, 2024-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc, 2.1.1, 2.0.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04.5 LTS

### Additional system information

Intel® Core™ i7-4790K CPU @ 4.00GHz × 8
32,0 GiB
NVIDIA Corporation GP107 [GeForce GTX 1050 Ti]

### Printer

Creality K1C

### How to reproduce

Start app image while LAN connection is enabled and connected printer is online

### Actual results

Crashes with segfault immedialtely after launching the appimage (recent projects are showing for a short moment) 

### Expected results

No crash

### Project file & Debug log uploads

Oct 25 19:44:16 alex-ubuntu kernel: [26820.404243] orcaslicer_main[59964]: segfault at 0 ip 000071a899903116 sp 00007ffe26b1a5c8 error 4 in libwebkit2gtk-4.0.so.37.71.3[71a89924d000+3546000] likely on CPU 0 (core 0, socket 0)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

It's a new issue. Everything worked fine until a few days ago. I tested appimage versions down to 2.0.0, they are all having the same issue.

When the LAN connection in Ubuntu is disabled or the conntected printer is turned off, the issue disappears.

When the slicer is already running (I have to disable LAN connection or turn the printer off to get the appimage running), the crash also occurs when I select the device tab or when I try to change the printer's connection parameters (hitting "Test" in the connection settings works fine). Uploading the GCODE to the printer works.

broncosis, 2024-10-26:
I noticed this as well on linux Mint 21.3   after recent updates 
just trying to figure out what updates I did today   but 
on 2 machines  same problem in my system on intel and on ryzen system 
went all the way back to version 2.00 that seems to  work ok 2.11 starts up as well 

I got 2.2.0 to start as well by deleting the network plug in and it started up 

not sure that will help alex  but I have 3 machines bambu x1c , stealth changer, and a railcore 

if I can provide any other info let me know

alex999888777, 2024-10-26:
Hi @broncosis , thanks for response.

I tried to delete the network plugin buat that didn't change anything. I had already deactivated it in the Orca preferences beforehand anyway. The issue still applied for all versions down to 2.0.0.

But I found a workaround and I'm pretty sure it's a fluidd-thing. I forgot to mention initially that my K1C is rooted and I used fluidd in the printer connection params (:4408 as URL) in OrcaSlicer. Now I changed the host type from Klipper to CrealityPrint and removed the port in the URL, and everything is working again. No crash at startup or when selecting the device tab. 
I remember there was a fluidd update a few days ago, that migh have caused the problems.

For me it doesn't really matter, because I never use the device tab in Orca. There is no difference when uploading GCODE anyway. 

Nevertheless, developers should keep an eye on it because there is indeed a problem with fluidd.

ressu, 2024-10-26:
I had the same issue, I pulled the [latest build from actions](https://github.com/SoftFever/OrcaSlicer/actions/runs/11521970983) which seems to work as expected. I don't really know what the source of the issue is though.

broncosis, 2024-10-26:
Yeah I can't use any profiles other than my Bambu ones otherwise it crashes 

It was super late last night I will try to pull some longs from my 2 computers tonight 

⁣Get BlueMail for Android ​

broncosis, 2024-10-26:
I tried the nightly build last night that was 15 hours old I will see if I can find a newer one 

Thanks for the reply 

⁣Get BlueMail for Android ​

alex999888777, 2024-10-26:
I just tried the newest build (from 4 hours ago), but it doesn't change anything for me. Still crashing when Klipper/fluidd is used for print host upload.

To make it clear again: the crash disappears when I go to the "Print Host upload"-section and remove the fluidd-port (4408) from the URL and switch the host type to CrealityPrint. When the URL includes the port, Orca will immedeatly chrash when I hit the OK button in the connection settings, and will continue to crash at every startup.

I noticed that "Upload and print" doesn't work anymore with the new settings. The GCODE will be uploaded, but the print doesn't start. I have to start it manually via the fluidd web interface. It's far better than the crashes of course, but the additional step is a little annyoing anyway. 

I also realized that I now can't use any version of Orca prior to the new 2.2.0 without losing the ability to directly upload the GCODE to the printer, because the older versions don't contain the CrealityPrint-host type, and none of the other host types is working without 4408 port for me.

broncosis, 2024-10-26:
[debug_Sat_Oct_26_11_26_02_55407.log.txt](https://github.com/user-attachments/files/17531833/debug_Sat_Oct_26_11_26_02_55407.log.txt)
attached is my log I have also tried the latest version
 but I think this might be related to issue 7210  aswell

ressu, 2024-10-26:
I have to also revise my previous comment. On the second start even the new binary crashes.

Aside for that, I haven't been able to narrow down the root cause. In my case the crash happens after a moment in the welcome screen. So based on the fact that there are reports that uploading gcode crashes and that Orca crashes on startup, it is possible that we have 2 distinct issues here.

broncosis, 2024-10-27:
I am almost certain this is connected   to https://github.com/SoftFever/OrcaSlicer/issues/7210

ressu, 2024-10-28:
Finally I had some time to test this out and yes, this seems to be a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/7210. The crash happens if Orca is able to talk to the API, which matches what was described in the other issue.

As a workaround if you don't want to tinker with your config while things are being worked out, you can use something like firejail to prevent network access:
```
firejail --net=none --appimage=orca.appimage
```

broncosis, 2024-10-28:
> The workaround mentioned by @el-quique in #7210 using the moonraker port is working fine. Even "Upload and print" works like expected. You just have to do without the Device tab.

I can confirm this aswell

closed 2025-02-04 by github-actions[bot]

broncosis, 2025-02-04:
Still no change on this