OrcaSlicer/OrcaSlicer #7239: copy and paste the same variable layer height settings to multiple objects.

wonsuhna1, 2024-10-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When using the VARIABLE LAYER HEIGHT function, I want to apply the same to multiple objects, but this is not possible.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

1. it would be nice to have the ability to copy and paste the same variable layer height settings to multiple objects.

2. additionally, the Support cubic infill algorithm does not work properly when importing multiple objects into a single object, or merging multiple objects to slice them in order to equalize the variable layer height.

closed 2025-02-01 by github-actions[bot]

voxain, 2025-10-17:
Mind blowing that noone seems to have thought about this. important feature imo!

Art-Porto, 2026-10-26:
This is implemented in #16462. The variable layer height tool now works on the whole selection: Adaptive, Smooth, Reset and painting apply to all the selected objects, either with one shared profile (what the prime tower needs) or with a profile for each object. Objects no longer need to be assembled into one to share a profile.

If you'd like to try it, test builds for Windows, macOS and Linux are attached to the PR's checks: [How to download PR artifacts](https://www.orcaslicer.com/wiki/how_to_download_pr_artifacts). A comment on the PR saying whether it works with your project would help the review.