OrcaSlicer/OrcaSlicer #7242: AMS warning for CF/GF filament is confusing.

DARKFiB3R, 2024-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

OS: Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

Perform temperature test with PLA-CF via the AMS

### Actual results

Warning message seems to sugest that I should remove the hardend steel nozzle

### Expected results

2. The nozzle hardness required by these filaments is higher than the default nozzle provided with the printer. Please ensure that you have replaced the default nozzle with an upgraded Hardened Steel Nozzle. It is also recommened to install a Hardened Steel Extruder Gear Assembly, otherwise all of the above mentioned components will be worn down or dameged.

Please click............

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DARKFiB3R, 2024-12-30:
How can I disable this warning entirely?

dentex, 2025-03-19:
I see the warning n.2 when starting a print using some PETG-CF with an hardened steel 0.6 nozzle (Bambulab A1)... Is that normal? If I'm not missing something this is the correct nozzle to use.

dentex, 2025-03-19:
I found the correct setting, if anyone's interested...

[image]

At first I thought it was the one under "Device->Printer Parts" (that also gives a warning if you have different dimensions)

Phil-Barker, 2025-04-09:
I get this warning every time I print PLA-CF since v2
and there's no option as above to say the printer has a hardened steel nozzle (which it does as I upgraded)

[image]

closed 2025-07-16 by github-actions[bot]