OrcaSlicer/OrcaSlicer #7244: Appimage(s) fail to launch on Steam Deck

radhaz, 2024-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.2.0-rc

### Operating System (OS)

Linux

### OS Version

SteamOS 3.5.19

### Printer

Ender3 v3 SE

### How to reproduce

1. Download appimage "OrcaSlicer_Linux_V2.2.0-rc.AppImage"
2. Change appimage permissions to make it executable
3. Execute said appimage
4. Note nothing happens
5. Open terminal
6. Run appimage from terminal
7. Get error message
8. Download the other appimage " OrcaSlicer_Linux_Ubuntu2404_V2.2.0-rc.AppImage"
9. Repeat steps 3 through 7

### Actual results

Executing OrcaSlicer_Linux_V2.2.0-rc.AppImage results in this error

`./OrcaSlicer_Linux_V2.2.0-rc.AppImage
/tmp/.mount_OrcaSlgJbAFs/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
`
Executing OrcaSlicer_Linux_Ubuntu2404_V2.2.0-rc.AppImage results in this error

`./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-rc.AppImage
/tmp/.mount_OrcaSliPN3B6/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory
`

### Expected results

OrcaSlicer successfully launching

### Project file & Debug log uploads

App doesn't launch therefore no project or log files are generated to upload.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

While researching I've seen mention of a Flatpak was being worked on; however, I couldn't find any mention of it for the past six months. Is there somewhere I could follow to keep up to date on the Flatpak version?

el-quique, 2024-10-26:
Hi try this:
sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

And create "start_orca.sh"
#!/bin/bash
WEBKIT_DISABLE_DMABUF_RENDERER=1 /opt/OrcaSlicer/OrcaSlicer.AppImage

radhaz, 2024-10-27:
@el-quique Thanks for your suggested work around. It "might" work but unfortunately I'm trying to work within the immutable distro's ecosystem so I haven't set the root passwd which will keep me from symbolically linking the files.

I already have OrcaSlicer working on my desktop, I would just like to be able to do some work with it "on the road" so this is more about me making the appropriate issue for the tracker and hope it gets acknowledged and/or finding where I can follow the news on the Flatpak development.

If you don't mind me asking I don't use a lot of appimages and I "think" I understand why I'd want to link 4.0.37 to 4.1.0 but what's the reasoning for linking the javascriptcoregtk(s) and also where would I learn more about the launch conditions for OrcaSlicer like WEBKIT_DISABLE_DMABUF_RENDERER=1?  If you just want to share a link to some documentation I could read that would be fine, I'm happy to learn. 

Again thanks for the response.

LinkerHoden, 2024-10-30:
I do have the same problem. Linking does not work since the libraries are not included in Steam OS.

anarsoul, 2024-11-01:
Flatpak for 2.2.0 can be found here: https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-1

radhaz, 2024-11-02:
> Flatpak for 2.2.0 can be found here: https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-1

Thank you so much for sharing the link! Now I can follow it's progress.

I'm going to close this issue not because the main problem is resolved but because I searched for just "libwebkit2gtk" and found multiple reports of it being "missing" from the appimage as far back as [18+ months ago](https://github.com/SoftFever/OrcaSlicer/issues/425) .  

In light fo this it probably isn't a SteamOS specific issue and more one with appimage so this post is not helping as there's already multiple issues addressing/referencing that.

closed 2024-11-02 by radhaz

dogs-1985, 2024-11-03:
> Thank you so much for sharing the link! Now I can follow it's progress.
>
> I'm going to close this issue not because the main problem is resolved but because I searched for just "libwebkit2gtk" and [...]

Hi, 

my understanding is that AppImage applications typically bundle most of the necessary libraries together so they can run on various system environments without relying on system-installed libraries. Is `libwebkit2gtk` an exception? Are such libraries usually expected to be installed on the system?

radhaz, 2024-11-03:
> Hi,
>
> my understanding is that AppImage applications typically bundle most of the necessary libraries together so they can run on various system environments without relying on system-installed [...]

You can skip most of what I wrote and just read [this post](https://phoenixnap.com/kb/flatpak-vs-snap-vs-appimage) that explains most of it better than I ever could.

I'm not experienced or knowledgeable enough to answer this definitively but it's my understanding that appimages by design are meant to be self contained applications packaged with everything to run on most distros. The size of appimages is smaller than other methods (snap, flatpak, etc) because it relies on the users OS to have "typical" packages already installed and if the user doesn't have them installed then they can install them to have interoperability with said appimage. 

I'm not sure the dev needs to roll every pkg requirement into the appimage because it seems to work fine for most distros and for the ones it doesn't most of those users can install the requirements. SteamOS whether we like it or not is immutable (as are a few other "boutique" distros) which brings inflexibility for the trade-off of stability, this means the appimage is not going to work on the deck unless the appimage maintainer included the requisite missing libs. Once the flatpak gets stood up and added to flathub this is going to become a non-issue because flatpaks DO come with all the required packages baked in. SteamOS is an outlier here and it's unreasonable for me to expect the dev to add in the libwebkit2gtk support since there's already two different appimages which I'm guessing is meant to support almost all the different distros out there.

Some people like the simplicity of Flatpaks being a "one-stop shop" whereas other folks take issue with the "bloat" that comes with installing multiple Flatpaks. It all comes down to the best thing in Linux the freedom to choose.

dogs-1985, 2024-11-03:
> You can skip most of what I wrote and just read [this post](https://phoenixnap.com/kb/flatpak-vs-snap-vs-appimage) that explains most of it better than I ever could.
>
> I'm not experienced or [...]

Thank you for the detailed explanation! However, I believe that the need for AppImage/Flatpak isn't limited to immutable systems. Many users might not have administrator privileges to install additional libraries, which makes AppImage appealing due to its portability and the ability to run on various Linux systems without the need for extra installations, even if the file size is larger.

AppImage gives the impression of being a portable solution that can be easily deployed on any Linux system, providing flexibility and ease of use. This is especially helpful for users who prefer not to or can't install extra system libraries.

I also don't mind having a AppImage version that includes minimal libraries alongside another version that has all necessary libraries bundled. This flexibility can cater to different user needs and system environments.

So, I would appreciate if this issue remains open and gets assigned to a developer for further consideration.

Thanks again for your insights!

radhaz, 2024-11-03:
> .....
> 
> AppImage gives the impression of being a portable solution that can be easily deployed on any Linux system, providing flexibility and ease of use. This is especially helpful for users who prefer not to or can't install extra system libraries.
> 
> ......
> 
> So, I would appreciate if this issue remains open and gets assigned to a developer for further consideration.

I closed this issue because I went through the open issue log and found that the issue I had wasn't relevant to the steam deck so much as the appimage and I found [several](https://github.com/SoftFever/OrcaSlicer/issues/6800) [other](https://github.com/SoftFever/OrcaSlicer/issues/6365) [different](https://github.com/SoftFever/OrcaSlicer/issues/5635) open [issues](https://github.com/SoftFever/OrcaSlicer/issues/425) about this already existing.

Your desire for appimages to be a one stop shop for everyone is a contentious topic that generates a lot of discussions online. I think most folks understand that appimages are going to work for most of the people most of the time and generally have the least bloat whereas Flatpaks are going to bring the "everything and the kitchen sink" approach to ensure the app operates properly in the sandboxed environment.

So rather than re-open this issue I think you would probably be best served tracking one of the other open issues regarding the missing dependency.