OrcaSlicer/OrcaSlicer #7249: Smarter Reduce infill retraction

jb510, 2024-10-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I print a lot of dust collection adapters, which are essentially 20-100mm pipes of various sorts.  I had horrible failures with stringing and blobs on the interior and couldn't figure out for days what was causing them. I was chasing temps, speeds, pressure advance and retraction.  I had not idea the "other" tab had a little checkbox called "reduce infill retraction" that appears to be on by default.

Disabling that eliminated 99% of the problems.  I'm perfectly happy with it always off and think that should be the default, but...

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would also seem to me that Orca could be made smart enough to detect the difference between a small or fully enclosed interior void, and an open interior void like a pipe would have.  That could make that setting a lot more useful to leave on. 

### Describe alternatives you've considered

1. defaulting "reduce infill retraction" to off
2. allowing users to set the reduced inflll retraction to a number/override.

jb510, 2024-10-26:
I'm going to do more testing. This could have been related to scarf joint as much if not more than infill retraction.

vgdh, 2024-10-27:
Scarf can leave some filament on the nozzle and it can cause stringing. 
You can try to use different print order like outer>inner. 
Also enable feature "avoid crossing walls"

closed 2024-10-29 by jb510

jb510, 2024-10-31:
FWIW, I think this was actually the problem described here #7191 and hopefully resolved by this patch #7208.