OrcaSlicer/OrcaSlicer #7251: Creality K1 printer does not appear when creating custom filament

gwmstram, 2024-10-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality K1

### How to reproduce

With 2 printers istalled Creality K1 and Ender-3 V2

1. With the K1 Printer Active go to Set Filaments to use (icon) > Custom Filaments > Create New
2. Opens "Create Filament" dialog 
3.  Enter Vendor "Ovature" (or any), Type "PLA" (or any) Serial "any"
4. In "Add Filament Preset Under" section any permutation of selections, exp I used: "Copy Current Filament Preset"
5. Only the Ender-3 V2 Appears as a printer choice. 

 

### Actual results

[image]

### Expected results

I expected to see the K1 as a choice for printer.  This has worked in the past.

### Project file & Debug log uploads

[debug_Sat_Oct_26_14_36_22_20408.log](https://github.com/user-attachments/files/17531999/debug_Sat_Oct_26_14_36_22_20408.log)
There is no project open. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I did a new install of 2.2.1 on a machine that never had this installed and got the same problem.
Problem occurs without a project being opened.

gwmstram, 2024-10-26:
After posting this, I added the K1C printer and it worked OK.   I added the K1 Max and it FAILED the same way as above.

gwmstram, 2024-10-28:
I installed v2.1.0 and no joy.   I installed v2.0 and IT WORKs.  Seems like something broke with v2.1

tkunchick, 2024-12-01:
same thing. I have already created some custom presets for my k1 Max, but trying to make a new one does not list the K1 Max
[image]
[image]

crosswick, 2025-01-03:
I have the same issue... stuck with 2.0.0 for now...

closed 2025-04-14 by github-actions[bot]