OrcaSlicer/OrcaSlicer #7252: Remove 36 plate limitation

laukejas, 2024-10-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently, OrcaSlicer limits a project to 36 plates max, a limitation inherited from BambuSlicer. This can be a problem when estimating time and material consumption for very large projects, as well as printing them and keeping things organized, requiring inconvenient workflows, such as removing some parts from the plates and adding others in their place to work around this limitation. For example, I have a project with 150 parts, where each part needs to be printed separately, meaning 150 different plates, and it is very difficult to properly plan it out.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice if there was no plate limitation, or the limit would be configurable by the user (based on their computer specs, etc.).

laukejas, 2025-01-25:
Bumping the issue due to risk of it becoming stale.

niklasb22, 2025-08-05:
@laukejas I tested increasing the max plate count to 255 and it works fine without technical issues.
That said, with that many plates the user experience gets pretty clunky.
I’d suggest bumping the limit to something like 64 plates instead — still a big improvement, but easier to manage in practice.

laukejas, 2025-08-05:
@niklasb22 Thank you for testing this! Why not make it unlimited, or at least 255 though? I get it, it gets clunky, but it's up to the user to deal with it or no. I don't see any advantage to limiting it to 64 plates. After all, user will only notice the clunkiness if they manually add that many plates, which I suspect will be pretty rare, so it won't affect casual users.

tina-junold, 2025-09-15:
I ran into this today while planning the T250. Hope this will be changed in future?