OrcaSlicer/OrcaSlicer #7263: Add support for horizontal and XY hole compensation with multicolor mode

WimVDK, 2024-10-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Certain projects i'm currently doing require multicolor mode and use a peg & hole type connection system, in single color mode that's not an issue as one can use use xy hole compensation and contour expansion. In multicolor mode thats not possible and the slicer gives a warning.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add support for multicolor mode AND XY hole compensation and XY Contour compensation

### Describe alternatives you've considered

There are no alternatives

spacenavy90, 2025-01-03:
Hoping this feature can be implemented.

8o8aua, 2025-02-26:
I'm waiting this feature also. This definitely needs to be done. 
More over setting up XY hole compensation need to be done on object section as well to override setting on global section of process profile