OrcaSlicer/OrcaSlicer #7264: QIDI Thumbnails

pmbroth, 2024-10-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

Ryzen 9, 40gb ram

### Printer

QIDI X-Max3

### How to reproduce

Send job to printer, to print.  thumbnails on screen on printer aree black

### Actual results

Send job to printer, to print.  thumbnails on screen on printer aree black

### Expected results

To see thumbnail of what is being printed.  I attached a simple image using two primatives.

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/17543625/logs.zip)
[job.zip](https://github.com/user-attachments/files/17543635/job.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Any file that is sent does not showe thumbnails

Arturis, 2024-10-30:
Experiencing the same issue. According to this previously reported stale bug ( [https://github.com/SoftFever/OrcaSlicer/issues/5973](https://github.com/SoftFever/OrcaSlicer/issues/5973) ) it is related to OrcaSlicer putting the images in a different place within the gcode than where Qidi is looking for them. It would be nice to either have this corrected automatically or to add a toggle for placing the image at the start or the end of the gcode. It is not acceptable to assume a user will have the know-how to download and compile a python script to get things to work correctly.

emudem, 2024-11-02:
I have firmware v4.2.15 and not working thumbnail in QIDI X PLUS  3

closed 2025-04-16 by github-actions[bot]