OrcaSlicer/OrcaSlicer #7269: M73 checkbox value ignored in Printer settings

tomasz-drgas, 2024-10-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Artillery Genius Marlin 2.x

### How to reproduce

1. Check the "Disable set remaining print time"
2. Slice a model and check if gcode contains M73

### Actual results

M73 is always in the gcode, the value of the "Disable set remaining print time" checkbox is ignored

### Expected results

M73 should not be outputed in the gcode file if "Disable set remaining print time" field is checked

### Project file & Debug log uploads

[NewLatch-4.5mm.zip](https://github.com/user-attachments/files/17546313/NewLatch-4.5mm.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I updated from v2.0.0 to v2.2.0 and the slicer started to output M73, even though the "Disable set remaining print time" field is checked. I tried to uncheck it, check it again and save printer's settings, but that didn't solve the issue. I rolled back to v2.1.1 and M73 is no longer added in the gcode (works as expected). I downloaded the portable version of v2.2.0 and I could replicate the issue again

EagleB3, 2024-10-30:
Windows 10 x64, printer with RepRapFirmware (mainboard Duet 3 6HC).
I confirm the presence of the described defect

Otscho, 2024-11-02:
Windows 11,  Marlin Firmware Sidewinder X1. Same Problem

fausby, 2024-11-06:
Same problem, but if the checkbox is checked, line M73 appears duplicated.

ghoethe, 2024-11-27:
no update for the M73 check box? still doesn't work :(

xlbt, 2024-11-27:
Hello,
I recently discovered OrcaSlicer, seems promising, but during my tests I noticed this annoying problem.
Two M73 lines are added regularly and when I check the “Disable set remaining print time” box, there's still one left.
Linux OrcaSlicer v2.2.0 on Artilery Sidewinder X1 with Marlin 2.1.2.1
Thanks

tomasz-drgas, 2024-11-27:
> Hello, I recently discovered OrcaSlicer, seems promising, but during my tests I noticed this annoying problem. Two M73 lines are added regularly and when I check the “Disable set remaining print [...]

As a workaround you have to downgrade to v2.1.1, hopefully this will get fixed soon.

xlbt, 2024-11-27:
> As a workaround you have to downgrade to v2.1.1, hopefully this will get fixed soon.

Unfortunately, it doesn't seem to have a ubuntu version of OrcaSliver 2.1.1.
Fingers crossed for a quick resolution.

Africatuin, 2024-11-29:
I have the same issue when I disable the time option. In the gcode the M73 commands are still present even if I select the checkbox. Artyllery X1-Windows 10

fausby, 2024-12-16:
Hello. My own temporary solution for Windows.

[removeM73.zip](https://github.com/user-attachments/files/18147539/removeM73.zip)

You have to add this batch to post-processing script.

[image]

duanecr, 2025-02-03:
This is still a problem for me.  Is anyone working on fixing this in OrcaSlicer, or will we just have to continue with workarounds?  This causes my printer to do a filament swap every time it sees this.

Schildkroet, 2025-02-12:
Please check/test https://github.com/SoftFever/OrcaSlicer/pull/8388

closed 2025-02-16 by SoftFever