OrcaSlicer/OrcaSlicer #7272: Allow mouse button mapping in preferences

julie777, 2024-10-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I use many software applications that require manipulation of objects. They all use the mouse buttons differently.
  - google maps uses the left button to pan
  - Fusion 360 uses the left button to select and move, the middle button to pan and the right button to orbit (with it set to mimic tinkercad)
  - Orca Slicer uses the left button to select and move objects, but it also uses the left button to orbit. The middle button and the right button both pan.
 
There is other software that I use that also causes problems with my muscle memory. Changing Fusion to the tinkercad setting fixed all my issues except the left mouse pan on maps. I frequently try to use the middle button to pan google maps (because I use Fusion so much.)

I am always trying to click and drag with the left button to select multiple items in Orca Slicer and I end up orbiting like crazy. Then I end up using `<ctrl>1` to fix the view.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to be able to map the mouse button actions when focused on the canvas to match Fusion. This would make it much easier to switch back and forth between Fusion and Orca Slicer.

Left click on object - select object
left click on empty space - clear selection and close pop-ups such as measure
left click and drag - move object
left click on empty space and drag - lasso select objects
right click on object - context menu
right click on empty space - orbit
middle click - pan

In addition, another nice feature of lasso select in Fusion is dragging left selects all objects touched by lasso, but dragging right only selects objects completely contained by lasso.

Orca Slicer already has a list of shortcuts for Prepare. It could just provide a dropdown to allow choosing the desired key.

BTW, the keyboard shortcuts/global has `<ctrl><tab>` wrong. It says "switch table page", but it actually just opens the next tab (ex: prepare to preview) and `<ctrl><shift><tab>` moves left one tab. This is nice because it works just like the browser, which is always a good choice give the ubiquity of the browser.

tlhintoq, 2024-10-29:
The fact that this is not the only ticket for fixing mouse behavior should be an indication of just how much people don't like the 
`Left-click on open space and orbit`
behavior - plus several other enhancements.

Its just WAY too easy to mis-click on an object, thinking you're moving it, and rapid orbit the entire world view.

Most of what the OP suggests is how Orca works now - minus lasso selection.
As an _incremental_ step, just moving orbit from `left-drag` to `right-drag` would make a lot of people really really happy.

buzzhuzz, 2024-10-31:
Looks like a logical improvement to a "Navigation Style" setting added some time ago. Currently there are default and touchpad (inspired by FreeCAD) navigation styles and we should be able to add  other mouse control styles too.

metalman1884, 2024-11-13:
Would be awesome to implement assignable mouse buttons to this already awesome slicer. How do we get this rolling to be included in the next release?? https://github.com/SoftFever/OrcaSlicer/issues/4440
Thanks to all capable programmers out there who spend their own time to improve something we all use, you're the best!

adoyle-h, 2025-02-15:
Remove stale label.

r3dbaba, 2025-09-15:
https://github.com/SoftFever/OrcaSlicer/pull/10736 allows assigning buttons. Everything except "lasso select objects" should now  be possible.