OrcaSlicer/OrcaSlicer #7273: Prepare viewport is totaly black

SEALBG, 2024-10-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Version: 23H2

### Additional system information

Processor	AMD Ryzen 7 4800H with Radeon Graphics            2.90 GHz
Installed RAM	32,0 GB (31,4 GB usable)
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 11 Pro
Version	23H2
Installed on	‎15.‎1.‎2024 ‎г.
OS build	22631.4391
Experience	Windows Feature Experience Pack 1000.22700.1047.0

### Printer

Bambu P1P

### How to reproduce

In the Prepare menu the viewport is black and can't see the bed representation. The "Reset Window Layout" does not make any change and can't find if there is any option that can repair that issue.

[image]

### Actual results

Black screen viewport in the Prepare is all black

### Expected results

The bed should be visible and to be able to add parts on it. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17547963/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

No need to open any project file. The black screen viewport is showing even in the main menu. I tried uninstalling the app and then reinstalling it, but the issue was not resolved. I'm not sure if the problem might be because of the video card drivers, but they are the latest version.

SEALBG, 2024-10-28:
Ignore the this! The problem has been detected and is expressed in the game profile of the video card!

tlhintoq, 2024-10-29:
@SEALBG   You should be able to close it with the button at the bottom of the thread, since you are the creator.

justindail311, 2024-10-30:
Confirmed on driver 24.10.1 and orcaslicer 2.2.0. Solution is in radeon software gaming tab then the graphics sub-tab and change profiles. Custom was an issue for me but performance mode cleared up the issue.

closed 2025-02-05 by github-actions[bot]