OrcaSlicer/OrcaSlicer #7275: VFA

Beep6581, 2024-10-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Manjaro

### Printer

n/a

### How to reproduce

Calibration > More > VFA

Rename "VFA" to a full and descriptive title. There are no bonus points for making the test name as short and obscure as possible.

The GitHub wiki page on calibration also lacks an entry on "VFA".

### Actual results

sdf

### Expected results

sdf

### Project file & Debug log uploads

jhygfhj

### Checklist of files to include

- [ ] Log file
- [ ] Project file

daef, 2025-01-27:
behave you darn bot - i came here to find out what vfa means...

Beep6581, 2025-01-27:
Beats me why anyone would enable this bot in their repo that just annoyingly spams and adds no value.

Orca's menu Calibration > Tutorial leads to a GitHub page which is based on or references to Andrew Ellis tutorials, and there VFA is defined as Vertical Fine Artifacts. Also known as MRR - Motor Resonance Rippling.

https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/vfas.html
https://www.obico.io/blog/speed-test-in-orcaslicer-vfa-a-comprehensive-guide/#the-vfa-test-in-orcaslicer
https://prorifi3d.com/pages/technical-details

Beep6581, 2025-05-02:
Lame bot.

ianalexis, 2025-08-19:
What do you think about this:
https://github.com/SoftFever/OrcaSlicer/wiki/vfa-calib ?

ianalexis, 2025-11-23:
> [@ianalexis](https://github.com/ianalexis) looks good.

Excelent, please close this issue now.