OrcaSlicer/OrcaSlicer #7276: Cursor disappears after using touchscreen

conte233, 2024-10-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

Windows

### OS Version

Windows 11 

### Printer

Prusa mk3s

### How to reproduce

Open Orca Slicer using touch
Interact with program using touch-only
Try to move mouse with trackpad 

### Actual results

The mouse cursor is not visible on the screen

### Expected results

The mouse cursor appears as expected

### Project file & Debug log uploads

[error]	2024-10-28 19:40:36.146936[Thread 0x0000297c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\Bat-Cat-Easy (1).stl
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\Pixel_Stand.stl
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\door_opener.stl
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\E-mount_rear_lenscap_Text_on_back.STL
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\SD_Top.stl
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\57mm Portafilter Funnel v1.step
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\Deckel.stl
[error]	2024-10-28 19:40:36.150433[Thread 0x00003060]:File Metadata/plate_1.png not found from archive
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\Part Studio 1.step
[error]	2024-10-28 19:40:36.150433[Thread 0x00003060]:Unable to open the fileC:\Users\conte\Downloads\10mm Spacer - 10mm Spacer.step
[error]	2024-10-28 19:40:36.150433[Thread 0x0000297c]:Unable to open the fileC:\Users\conte\Downloads\57mm Portafilter Funnel - v2.step
[error]	2024-10-28 19:40:41.046873[Thread 0x0000297c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-10-28 19:40:41.159470[Thread 0x0000297c]:Slic3r::NetworkAgent::get_version, get_version not supported,return 00.00.00.00!
[error]	2024-10-28 19:40:42.997052[Thread 0x0000297c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SCiunczyk, 2025-01-13:
I have this same problem on my Surface Book (Windows 11).

bobbyb2024, 2025-02-16:
I'm experiencing this same problem on a Asus Zenbook Flip, running windows 11

kotarokun, 2025-05-21:
same problem with Surface pro 6... can't see my mouse

sutty86, 2025-07-28:
Also having same issue with Rog ally

64Bit-Watermelone, 2026-02-14:
Also having this issue on my Lenovo yoga 6 win11

yinxistats, 2026-07-04:
same issue with touch screen asus swift go 14