OrcaSlicer/OrcaSlicer #7282: Vertical Brick Layers/Interlock Layers

theboee, 2024-10-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am reopening this enhancement since the old one, #4133, was closed and marked stale a while back. I am doing this to try to get contributors attention and hopefully get this in the slicer. I expect this to be closed since it is a duplicate, so no hard feelings on that front.

[image]

This process results in stronger interlayer bonds and in turn increases layer adhesion in the Z plane. Initial testing by CNCKITCHEN had the improvement in strength somewhere around 10-14%, but he reveals that this is with "old wet and unreliable filament.". The true numbers could be higher when used in conjunction with good dry filament.

https://www.youtube.com/watch?v=5hGm6cubFVs

[image]
[image]
[image]

This would help all functional prints across every printer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

There should be an option to enable "Brick Layers" or, for possible rights issues reasons, something like "Interlock Layers".

### Describe alternatives you've considered

Learn how to code myself but I don't really have the time or talent so...

theboee, 2024-11-01:
@SoftFever

theboee, 2024-11-01:
Another thing to add, would be to increase the flow rate and line width for the offset layers to help aid adhesion.

If your material requires a 1.02 Flow Rate, than you'd tune your functional parts with Vertical Brick Layers to over extrude on the offset layers. This would get rid of the gaps that formed when CNCKitchen tested this.

[image]

So when working with this, the slicer settings could look like -

Offset Extrusion Multiplier/ 1.0 or % of existing Flow Rate

Offset Extrusion Width/ 0.4mm or % of default line width

theboee, 2024-11-01:
> I made a change a patch which adds layer interlocking by making intersection look like brick layers (like actual brick wall with layers shifted in horizontal direction): you could read about it here #5261
> 
> I've since found and resolved a root cause of the issue which made me develop this patch (under-extrusion on acceleration), but I still always use it for all my engineering prints.
> 
> It should be possible to improve that change for merging into the code base, but I'm not sure if I would be able to fix walls finishing quality.

This is also an interesting way of aiding z strength. I wonder if there may be a way of combining these methods?

VigilantGardener, 2024-11-07:
> Another thing to add, would be to increase the flow rate and line width for the offset layers to help aid adhesion.
> 
> If your material requires a 1.02 Flow Rate, than you'd tune your functional parts with Vertical Brick Layers to over extrude on the offset layers. This would get rid of the gaps that formed when CNCKitchen tested this.
> 
> ![8875471](https://private-user-images.githubusercontent.com/73860654/382187162-f5c54871-692a-464a-a6b8-ce70a57758ee.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5ODE3NzYsIm5iZiI6MTczMDk4MTQ3NiwicGF0aCI6Ii83Mzg2MDY1NC8zODIxODcxNjItZjVjNTQ4NzEtNjkyYS00NjRhLWE2YjgtY2U3MGE1Nzc1OGVlLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA3VDEyMTExNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3MmUwZGIyNzZhOTYzNGE2ZTQ4ZDM5ZDM2MGEwYzI0NzUzOWY3MDllNzJhNDZjZjUwNDkxMWYxMDQwZjc0MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.vM0KH4NImSWioYcgn3fw02TO_6aX7gxg8hdfLU4qwXo)
> 
> So when working with this, the slicer settings could look like -
> 
> Offset Extrusion Multiplier/ 1.0 or % of existing Flow Rate
> 
> Offset Extrusion Width/ 0.4mm or % of default line width

This would be a key facet of this feature, allowing the full gap to be filled with material.

Additionally, the ability to control the part cooling fan for the offset layers.  The offset layers shouldn't be part of critical part geometry, they're just about strength.  The ability to alter part cooling, to include no part cooling, would mean that filament could go down hot for maximum inter-wall and inter-layer adhesion.

rkarlsba, 2024-11-08:
It seems tis is patented, but that patent is fake - it just refers to common knowledge https://www.youtube.com/watch?v=9IdNA_hWiyE

VigilantGardener, 2024-11-08:
> It seems tis is patented, but that patent is fake - it just refers to common knowledge https://www.youtube.com/watch?v=9IdNA_hWiyE

Which is a bummer, but it seems like the 1995 patent would hold sway and I think legally the feature can be implemented in any slicer now.

SQalliT, 2024-11-09:
> > It seems tis is patented, but that patent is fake - it just refers to common knowledge https://www.youtube.com/watch?v=9IdNA_hWiyE
> 
> Which is a bummer, but it seems like the 1995 patent would hold sway and I think legally the feature can be implemented in any slicer now.

The video states that the patent is expired in the US and EU now, so anybody can print this way now.

kmetani, 2024-11-09:
I would love to try this, where can i get the necessary files to try this on my own slicer and printer, please note that I've never tried plug ins on any slicer before so, neither used github much in my life so use simple language please.

kmetani, 2024-11-09:
This is great! I would love to try this on my own slicer software and 3d print using it, how can I download the necessary files to use your method of brick layering. Please note that I barely use github and can't get my way in it very well, so use easy language for the instructions on downloading it please.

weaseljohnson, 2024-11-10:
Commenting here for visibility, and to reiterate the point that this feature is no longer patented. 

https://youtu.be/9IdNA_hWiyE

rkarlsba, 2024-11-10:
> Commenting here for visibility, and to reiterate the point that this feature is no longer patented.
> 
> https://youtu.be/9IdNA_hWiyE

I posted that above ;)

scottdky, 2024-11-10:
All three techniques should be implemented: 
1. Staggered layers in z direction 
2. Higher flow rate on inside perimeters 
3. Stop fan on inside perimeters 

By implementing all three would result in even greater bonding than the 10 to 15% from just item 1. It would be very close to isotropic.

robertjtaggartjr, 2024-12-05:
Yes please implement this  !!!!!

mstrblueskys, 2024-12-07:
> Additionally, the ability to control the part cooling fan for the offset layers. The offset layers shouldn't be part of critical part geometry, they're just about strength. The ability to alter part cooling, to include no part cooling, would mean that filament could go down hot for maximum inter-wall and inter-layer adhesion.
> 
> EDIT: Flow rate Cooling Speed
> 
> Minimum available settings to leverage full value from this feature

The hard part about that would be inner walls are higher for half the print, so you might get seepage over the outer walls.

I wonder if the answer is not offsetting layers, but halving the inner wall layer height and printing one that's at a materially significant increase in temp without cooling to allow it to seep and then covering it up with a regular layer with normal cooling. If we're trying to increase surface area between layers, this might give us more. See my awful paint skills below. The outer layers print normal, the green layer is an hot, oozy one, and the red layer is normal again.

[image]
 
Maybe this would get around that awful "patent" that's out there, too.

Sab-tech-lab, 2024-12-12:
There is another idea on reddit:
https://www.reddit.com/r/3Dprinting/comments/1fp65o5/idea_advanced_method_of_brick_layering_for_fdm/

robingosse, 2024-12-15:
Following...

I desperately want this feature implimented in Orcaslicer. Is anyone working on it? I've been doing some research, I've read about the patent shenannigans, I've seen some clever work arounds where the printer makes a "moving centipede" shape with a single bead, I'm not sure how else to describe it, so it interlocks in 3 directions rather than 2D layers. (link here: https://www.reddit.com/r/3Dprinting/comments/1fp65o5/idea_advanced_method_of_brick_layering_for_fdm/?share_id=p-h3mZitRuujvYZZ0Kh_t&utm_content=1&utm_medium=android_app&utm_name=androidcss&utm_source=share&utm_term=1)

I just want this feature. Anyone have any recent updates?

Anthony-Bec, 2024-12-16:
following

tilmaen, 2025-01-11:
I posted the following in the cura github already:

I would totally love this feature.
The european Patent hasn't been awarded yet (4 years in limbo). it seems it is stuck because of a more thorough examination process. 3rd-Party observations have been filed (and acknowledged), also the examiner seemed to have found several issues of "non novelty", Also the bungled up references to the prior Patents were critizized by the EPC officer (as opposed to the USTPO).
The correspondence is pretty interesting [https://register.epo.org/application?number=EP21175830&lng=en&tab=doclist](https://github.com/Ultimaker/Cura/issues/url)
i have a friend who works there (different department) i might ask him, whether he can tell me something about that patent application.

Charmio1, 2025-01-17:
@tilmaen Woah!! What a great resource. That definitely was worth a read.
From that official website, it seems like quite an active discussion is going on right now over at the European Painted Office.

Lots of "Non-patent literature filed by a third party" going on very recently. 
I wonder who our third party knight in shining armor is?

From my interpretation it seems open and shut that the patent is invalid.

(FYI, there's an issue with how you posted that link, it takes you to an empty page on the cura git.) 
[European Patent Register: EP4094924](https://register.epo.org/application?number=EP21175830&lng=en&tab=doclist)

Gridzilla, 2025-01-17:
I would love to see this implemented

nubnubbud, 2025-01-22:
IANAL/TINLA
I posted a response to the US patent issue here; https://github.com/SoftFever/OrcaSlicer/issues/7499#issuecomment-2606221367 
but by and large, I really don't think it's enforceable. best case, nothing happens. worst case, we obey a cease and desist and challenge the patent. The US has financial services for people and organizations who need assistance with patent work, and while it probably won't be totally free, all the legal fees can be handled by them if someone like me were to file a petition...

but the most important part is that the patent holder decides whether or not to pursue perceived violations, and it is quite common (and perhaps the most straightforward way, given US bureaucracy) to invalidate a patent in this way. The USTPO handles far more patents than european ones, and I expect their standards have to be lax to be able to award these patents within years instead of decades.
Specifically:
A patent with incorrect references may not necessarily be completely invalid, but it can be significantly weakened and more vulnerable to challenges in court, especially if the incorrect references are material to the patentability of the invention and were not disclosed to the patent office during examination; in such cases, the patent could be invalidated if the missing or incorrect references demonstrate that the invention is obvious or already publicly known.

and in this new patent's case, one reference is... not right, the other is incorrect, and if the patent office had seen the correct one, would have seen it was a copy of an expired one, claiming it builds on it by... mentioning strength. when the original one also notes 

> "These fluid pockets or voids should be contained in the nominally solid portions of the parts. While such parts are less strong and dense than non-porous parts, they are more dimensionally accurate"
 and by implicit reasoning, the 1995 paper states that more porous parts are weaker, and less porous parts are stronger. as well...

> "In some cases the beads in adjacent horizontal layers will locally be parallel and laterally offset from each other by half a diameter. In this situation even stiff or rapidly solidifying materials will pack with a porosity of about 9%. Further, bead elements may be deposited in successive horizontal layers in a skew arrangement 50 as to provide a minimum porosity"

I believe this is stating that there are cases in which one side of the offset layers is parallel, the next layer is displaced, and begins the offset pattern, as shown in an image labeled "50" in its references. 

[image]

if this is true, then it means that the current patent has no novel features or considerations, even if you ignore existing public literature, that slicers are capable of it already (if in a tedious and generally inaccessible manner) and that other patent offices have found the errors in the paper contentious.

and to top it all off, the patent only exists in a completed state in the US. If the leader of this project is not in the US...
and behold, SoftFever is based in Singapore, none of this should be any issue for publishing in the first place, and the patent holder can't exactly hold a developer from singapore accountable for someone in the US breaking US patent law with open source software which is legal in singapore.

I wouldn't be worried. long-winded? definitely. afraid of Mr. Saberton? well, he might wanna change his name to Sabaton, knowing where the legal system is gonna bury its foot if it sees he made a fool of their patent system by providing the wrong references.

bo0tzz, 2025-01-22:
@TengerTechnologies just released a post-processing script that implements this: https://github.com/TengerTechnologies/Bricklayers

AtHeartEngineer, 2025-01-22:
@theboee / all, thanks for pushing this and being awesome

buzzhuzz, 2025-01-22:
I see no one talking about sloped walls.  One of the reasons, besides implementation complexity, I'm using horizontal interlock is really complex filament flow in case of sloped walls:

[image]

Because of this I'd like to agree requirements first, probably make some testing/prototyping first. Main problem here would be external wall finishing quality.

nubnubbud, 2025-01-23:
> I see no one talking about sloped walls. One of the reasons, besides implementation complexity, I'm using horizontal interlock is really complex filament flow in case of sloped walls:
>
> [image] [...]

[image]

the middle option is attractive, except that it involves wider layers, potentially wider than a standard nozzle can produce. I would err on the side of a thinner extrusion and full size extrusions, rather than normal extrusions and larger ones.

as well, orcaslicer can already identify overhangs, and arachne can be told to vary line widths for those areas. past 30* slope, it could transition to standard, because the normal function gives just about the same effect anyway, and if you're using the middle option, it should probably make thicker extrusions for support on overhangs, but thinner extrusions on upper angles to ensure the model is accurate. Not totally sure on whether larger or thinner extrusions are better at the moment.

**As a naive approach, I would begin by 
1- only considering the valleys between touching extrusions places to print. 
2- from that starting point, it should be easily modifiable to add thin perimeters to reduce "double-stepping" or thick perimeters, depending on which would be closest to the nominal nozzle size
3- consider in the case of vertical interlock filling in excessive gaps wwith thin perimeters, or leaving it a bit more open, and over-extruding into the valley.**

theboee, 2025-01-26:
Too anyone following this issue, it is now being implemented in slicer - #8181

Rebel98luder, 2025-02-13:
Can't wait for this feature!

tlhintoq, 2025-06-01:
Just sayin'.... 
If there's no conflicts and it *can* be merged... And you've got notable YouTubers like CNC Kitchen posting about using it...  Might be time to merge it.
https://www.youtube.com/watch?v=dDgA51zdfLc

igiannakas, 2025-06-01:
Both Softfever and I have open review comments in the code since feb hence need to be addressed or discussed but haven’t seen a reply since then.