OrcaSlicer/OrcaSlicer #7290: V2.2.0 crashes on startup

Wakinyantanka, 2024-10-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.2.0

### Operating System (OS)

Linux

### OS Version

Linux mint

### Printer

Custom BLVMGN Cube

### How to reproduce

Click icon software starts and immediately crashes 

### Actual results

Software crashes

### Expected results

Operational software that doesn't crash on startup

### Project file & Debug log uploads

Cant get log file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Can't get log file. program isn't open long enough

Schakal2102, 2024-10-29:
Ab Linux Mint 22, soll die [OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage] verwendet werden.
Bei mir läuft die andere auch nicht, aber diese Version ohne Probleme.

Wakinyantanka, 2024-10-29:
I'm sorry is there a way to translate this to English?

CitrusIntellect, 2024-10-30:
Translated bu DuckDuckGo:

Starting with Linux Mint 22, the [OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage] should be used. The other one doesn't work for me either, but this version without any problems.

Let me guess, your using an nvidia card 

Your not alone  #6908 #6520

CitrusIntellect, 2024-10-30:
#7210

Wakinyantanka, 2024-10-30:
Yes using nvidia gpu. I tried OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage and it won't even open. Did a fresh install of Mint with the same problems. Please tell me they are working on a fix for this. I do not want to go back to Windoze....

kohbo, 2024-10-30:
I have this issue and I'm using an AMD 7900xtx

kohbo, 2024-10-30:
Switching to the latest version on Github 2.2.0 beta 2 is working for me

padcom, 2024-10-30:
Not working for me when the selected printer is networked - aka the network plugin is enabled an in use. Linux Mint 21 + RTX3060ti if anyone is interested. The issue started happening Today. The network plugin got updated and boom - segfault.

klack, 2024-10-30:
It appears to be GTK errors for me:
```
(orca-slicer:159484): Gtk-CRITICAL **: 13:53:45.984: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:159484): Gtk-CRITICAL **: 13:53:45.984: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:159484): GLib-GObject-CRITICAL **: 13:53:45.984: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:159484): Gtk-CRITICAL **: 13:53:45.984: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:159484): Gtk-CRITICAL **: 13:53:45.985: gtk_widget_set_size_request: assertion 'width >= -1' failed
```

klack, 2024-10-30:
Deleting my config folder worked for me.  See:
See https://github.com/SoftFever/OrcaSlicer/issues/6887

Wakinyantanka, 2024-10-30:
I'll try the beta 2.2.0 when I get home.

CitrusIntellect, 2024-10-31:
I just compiled OrcaSlicer from nothing and it is working great! I would suggest that first.

darco415, 2024-10-31:
> I just compiled OrcaSlicer from nothing and it is working great! I would suggest that first.

can you share it?

Wakinyantanka, 2024-10-31:
> I just compiled OrcaSlicer from nothing and it is working great! I would suggest that first.

What is the process to do that? Thanks!

Wakinyantanka, 2024-10-31:
I followed the read.me for Ubuntu and get an error that stops the build..

darco415, 2024-11-01:
> ninja: build stopped: subcommand failed."

same here

CitrusIntellect, 2024-11-02:
Did you follow instructions for Ubuntu or Linux compilation?

Also if your on Arch Linux and have the aur you can run:
paru -Syu orca-slicer
OR
yay -Syu orca-slicer

Depends on your AUR helper

darco415, 2024-11-02:
> Deleting my config folder worked for me. See #6887

Ok so I renamed the machine folder, and OS started up fine. Able to upload file to printer

Wakinyantanka, 2024-11-02:
I purged everything from my machine then installed the Linux appimage,(Ubuntu appimage still won't open at all), rebuilt all my profiles for 6 machines and upload through mainsail. It's the only way I can get it to work. Every time I try to put in the IP address it immediately crashes and will not reopen.

Wakinyantanka, 2024-11-02:
> Did you follow instructions for Ubuntu or Linux compilation?
> 
> Also if your on Arch Linux and have the aur you can run: paru -Syu orca-slicer OR yay -Syu orca-slicer
> 
> Depends on your AUR helper

I followed The instructions for Ubuntu.

darco415, 2024-11-02:
> > Did you follow instructions for Ubuntu or Linux compilation?
> > Also if your on Arch Linux and have the aur you can run: paru -Syu orca-slicer OR yay -Syu orca-slicer
> > Depends on your AUR helper
> 
> I followed The instructions for Ubuntu. I'm using Linux Mint Cinnamon 21.3

add the port number 7125 after the ip

darco415, 2024-11-02:
> I purged everything from my machine then installed the Linux appimage,(Ubuntu appimage still won't open at all), rebuilt all my profiles for 6 machines and upload through mainsail. It's the only way [...]

add the port number 7125 after the ip

Wakinyantanka, 2024-11-02:
> 
> add the port number 7125 after the ip
Added the port number and I am able to upload directly to my machines without crashing. 
[image]
Device tab shows this though.
Still happy to be able to upload directly!!

henr7381, 2025-01-07:
I assume this is still an issue that is being addressed, but it's not fixed so I figured I'd post this. 

System details:

  Kernel: 5.15.0-130-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy

As of 20250107, the dedicated Ubuntu appimage does not launch (to be expected). Furthermore, the appimage crashes when trying to get ahold of the Klipper interface on my printers. I know this, because when I isolate my machine from the subnet with my printers, the appimage launches and runs fine. If I then reactivate the subnet and allow Orcaslicer to try and connect to my printers, it immediately crashes. Putting the Moonraker port after the IP of my printers prevents Orcaslicer from crashing, so there must be an issue with the Klipper interface tab in Orcaslicer.

If you're another user with this problem, as mentioned previously, add `xxx.xxx.xxx.xxx:7125` to the end of your printer's IP to prevent crashes.

darco415, 2025-01-09:
ya we known this fix since Nov 2

closed 2025-04-17 by github-actions[bot]