OrcaSlicer/OrcaSlicer #7292: Generic PLA shows as ABS

soundtkllc, 2024-10-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

All Bambu

### How to reproduce

When you set the filament type in the printer (x1c only) or in the slicer to generic pla it always shows it as ABS. you have to set the type to bambu basic pla to get around this

### Actual results

When you set the filament type in the printer (x1c only) or in the slicer to generic pla it always shows it as ABS. you have to set the type to bambu basic pla to get around this

### Expected results

it should show as pla not abs

### Project file & Debug log uploads

don't have

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Smerks101, 2024-11-01:
This is also occurring on the P1P

edgarbs1998, 2024-11-02:
Same issue here on an A1. However, for me it does not show ABS but ASA and it is stuck on it, I just can't change to Generic PLA. On Bambu Studio it is fine.

soundtkllc, 2024-11-02:
Originally I said ABS but after reading another person’s comment I confirmed it is actually ASA and I put incorrect info in original post when I stated it was ABS. Same problem but it shows the filament type loaded as ASA in the slicer device tab, when it is clearly entered as Generic PLA, instead of my above statement that it shows ABS.

Smerks101, 2024-11-03:
I uninstalled the newest version so I cannot check to confirm but it could very well have been ASA instead of ABS.

dartrax, 2024-11-16:
Same issue here with ABS instead of PLA. There is a good video showing the behavior in #7432.

Smerks101, 2024-11-17:
> Same issue here with ABS instead of PLA. There is a good video showing the behavior in #7432.

Ive since given up and went back to Bambu slicer. I posted a message about this in the discord and got grief for not posting it here.

dartrax, 2025-02-16:
unstale

soundtkllc, 2025-02-16:
Not stale because the issue has never been fixed or even addressed yet

dartrax, 2025-05-18:
@soundtkllc This may now be fixed in #4690, can you test it and give feedback there?

dartrax, 2025-08-17:
may be fixed in #4690

acotam2, 2026-02-15:
Still here on 2.3.2-beta, Generic PLA is shown as ASA, generic PETG is PET. Some other randomly checked filaments seem okay.