OrcaSlicer/OrcaSlicer #7300: Allow tree support to grow ontop of objects

lin-ycv, 2024-10-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I really like slim tree supports, they're very efficient and works most of the time.
The issue is that when the overhange is very far from the bed, the tree is not so efficient anymore, because the trunk needs to be from the bed.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It'd be nice if the tree support can act like normal snug supports, allowing it to grow from ontop of object instead of from the bed. So instead of having the green support in the screenshot below, it's grow the yellow area.
[image]

hotellonely, 2024-10-30:
problem with thin tree (and trees in general with too little wall layers) is that they snap easily (thin trees are very fragile), for some materials it's not even doable. sometimes even the default tree would fail that's why people also developed strong trees. 
https://www.youtube.com/watch?v=SHdh_JtnxWE
Check 3:08 of this video to see.

lin-ycv, 2024-10-30:
yes, tree support it's a golden bullet, but it works for me.
Even with tree strong, it still grows from the build plate, i feel that it should be able to grow from ontop of the object to save material, whether or not it fails...etc, I think that's model specific, not really a reason why trees can't grow from objects

hotellonely, 2024-10-30:
[image]
Also if you uncheck the "on build plate only" option, the tree support do grow THROUGH the model, which personally I don't agree with but

closed 2025-02-05 by github-actions[bot]