OrcaSlicer/OrcaSlicer #7305: More detailed filament estimation, in case of colour change

DVSVIDEO, 2024-10-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Yesterday I printed an object in which I wanted to add gold layers in 3 places (in different layers).
There was very little gold filament left.
I sliced the model, adjusted the colour swapping layers and sliced it again.
Unfortunately Orcaslicer does not show how much material is needed for the layers between the colour changes, so I ran out of gold and the model failed.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In case of Pause or M600, display the parameters (filament length, filament weight, price) of the layers between them separately and then in total.
see the photo below:

[image]

kylek29, 2024-10-31:
That does seem like a nifty feature of PrusaSlicer, hopefully someone can port it.

jdygert, 2024-11-07:
You can at least see length and weight by switching to the "Filament" color scheme:

[image]

Note that you must re-slice or switch to another tab and then back to the preview. (That might be a bug?)

DVSVIDEO, 2024-11-07:
you misunderstood!
What you are claiming is true if you have some sort of color swapping system and you set at least 2 different colors for the filaments.
What I'm talking about is the manual colour swapping used on single head, single colour (standard) printers when reaching a given layer.
By dragging the slider, inserting a color swap (M600 or PAUSE) command at the desired layer and reslicing, OrcaSlicer will not display a different color, will not calculate new color weight and length data.
PrusaSlicer has it but here this feature is completely missing:

[image]

DVSVIDEO, 2025-02-07:
Miért?