OrcaSlicer/OrcaSlicer #7307: Max Flow rate being ignored

convicte, 2024-10-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1.

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Ryzen 5950X
32GB DDR4

### Printer

Anycubic Vyper

### How to reproduce

Any max flow rate set is being completely ignored, allowing excessive speeds for the combination of layer height and deposition width.
The slicer indicates maintaining set max flow rate, while exceeding it at time 2 times.

Since it's been an issue reported for several years, across several tickets, auto-closed and unresolved, I hope it's not really necessary to provide much more detailed reproduction steps, but should this not be the case, I'd be happy to do so.

The interesting part however is that the old bug report would see the UI display incorrect flow rates:
#1216 and so would #2033 at first, but for 2.1.X this UI got fixed but the flow remains off as a result of speed vs. line width calculation.

The issue reported is similar to #7037 but the reporter indicates his problem being UI, while it most likely isn't.

### Actual results

[image]

[image]

https://makeit-3d.de/en/tools/maximum-printing-speed.html#:~:text=Determine%20maximum%20print%20speed,calculate%20the%20maximum%20printing%20speed.
[image]

The flow rate required to maintain 163mm/s at 0.25 layer height and 0.62 line width is roughly 25.4mm3/s
[image]
I've seen this discrepancy get even more excesive

### Expected results

Max flow rate being considered when slicing

### Project file & Debug log uploads

Built-in Orca Cube

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Zogar89, 2024-10-30:
I disagree. Some factors can change the real flow like Flow Ratio, arachne with variable line width etc..

convicte, 2024-10-30:
> I disagree. Some factors can change the real flow like Flow Ratio, arachne with variable line width etc..

By over 25% of the set value... and at time over 50%...

You are absolutely entitled to an opinion, though all my settings should decrease the expected flow rate, not increasing it. Both the calculations and my print effects (which I systematically tested with 5 Cura cubes) show the printer is attempting to push more material than the max flow set and what the device is capable of.

So TL:DR, the max flow is not enforced.

ShaneDelmore, 2024-10-31:
I use max flow rate to control all of my speeds daily and it works great.  Which is not to say you don't have an issue, only that it's not completely broken so there must be some set of circumstances where it breaks. I spent all morning today doing pressure advance testing at various flow rates and accels and orca estimate matched what klipper (mainsail UI) displayed within 5%.  I wonder what is triggering the behavior you are running into.

convicte, 2024-10-31:
If I curtail nothing but the speed, to prevent the printer from exceeding the flow cap when running calculations, I end up with the same cube printed very well.

I'd say heading towards perfection, not using ironing and ignoring the VFAs, which are unavoidable at 160-180mm/s on a bed slinger:
[image]
[image]

All the flow fine tuning and linear advances seem perfectly reasonable, so I arrive at max flow control as the only culprit and the UI seems to confirm it.

convicte, 2024-10-31:
[image]

[image]

[image]

dewi-ny-je, 2025-01-16:
I just tried to reproduce the bug and I found some spots where the max flow rate appears:

[image]

[image]

[image]

[image]

Please notice how in both moves NO extrusion is actually performed (there's not "E" component in the G1 command) and see how the value calculated in the bottom slider exceeds the max flow listed in the graph on the top-right pane.

I think that multiple bugs are present in the flow calculation code, for sure in the display part, not sure about actual printing. That would require parsing the gcode to calculate each actual extrusion rate

GriAle, 2025-02-28:
[

[image]

]

Have some problem

ShaneDelmore, 2025-03-06:
> I've decided to use the VFA Orca test as a quick and simple constant flow vs. constant speed test.
> 
> With 60-200 mm/s range I get no issues maintaining the print at speed, but in corners, things go completely pear shaped: [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> I've fine-tuned the max flow through line thickness and layer height, to prevent it from going outside what my hot end can manage: [image]
> 
> Not sure if it's more informative or more confusing, honestly... With your recent efforts on Pressure advance and the like, [@ShaneDelmore](https://github.com/ShaneDelmore) would you venture a guess what I may be dealing with here.

Just noticed this, did you ever figure it out?  That looks like too much pressure advance at first glance.

GriAle, 2025-03-12:
in 2.3.0-rc it's only gotten worse. 

[image]

onolox, 2025-03-26:
Somehow it worked, but now is completely ignored, strange, in PrusaSlicer this setting always worked fine.

ShantanuNair, 2025-03-31:
🎉
@convicte @stefansjs I figured it out.

**Issue**: The max volumetric speed is applied to features in such a way that it's not applied after flow ratio is factored in, and so the max volumetric flows for features are set, and then the flow ratio is applied bumping it up to beyond max flow set in Orca.

**To test**: Set your Flow Ratio for the filament to something less than 1.0, say 0.98, and it will respect your max volumetric flow limits. The further above 1.0 you go, the more you see flow rates crossing the max volumetric speed

### Solution: 
Say you're running a flow rate of **1.125**, and you want to set a max volumetric flow of **25mm3/s**:

_**Orca Max Volumetric Speed = Expected Max Flow Rate / Flow Ratio = 25 / 1.125 = 22.222**_

The setting of **22.22** will result in gcode that adheres to your max flow rate, and can be confirmed via the preview.

@SoftFever Is this expected, perhaps another box showing calculated max flow rate, or adjusting it to factor in flow ratio would make for a good PR? I haven't worked on slicer software ever, and just got into printing, but would be happy to contribute.

It seems that maybe to some (or as it should be), the ratio should be a last step multiplier, more of a fine tune that can cross the max volumetric speed limit. Imo, it would make sense to change it to actually enforce a max flow on the gcode, which it doesn't now.

dewi-ny-je, 2025-03-31:
> **To test**: Set your Flow Ratio for the filament to something less than 1.0, say 0.98, and it will respect your max volumetric flow limits. The further above 1.0 you go, the more you see flow rates crossing the max volumetric speed
> 
> ### Solution:
> Say you're running a flow ratio of **1.125**, and you want to set a max volumetric flow of **25mm3/s**:
> 
> _**Orca Max Volumetric Speed = Expected Max Flow Rate / Flow Ratio = 25 / 1.125 = 22.222**_
> 
> The setting of **22.22** will result in gcode that adheres to your max flow rate, and can be confirmed via the preview.
> 

@ShantanuNair no it doesn't work the way you say.

I loaded an Orca cube AND a Voron cube (because just one of the two makes the issue less evident, which already negates your conclusions), then I set flow to 0.95 as you suggested. Max flow is set to 14 mm3/s.
Classic wall generator gave me about 16 mm3/s as max flow rate, Arachne gives me 20 mm3/s actual max flow.

Playing with flow multiplier, I see this:

Classic, 1.01: 15 mm3/s
Arachne, 1.01: 21 mm3/s

Classic, 0.95: 16 mm3/s 
Arachne, 0.95: 21 mm3/s

I rounded, but for Arachne there is literally no difference down to the decimals.

The issue is definitely not what you seem to have described.
Provide a test file if you want us to confirm, this is what I used.

[flow test.zip](https://github.com/user-attachments/files/19535230/flow.test.zip)

ShantanuNair, 2025-03-31:
@dewi-ny-je I also tested the same with both the voron and orca and when flow rate set to <1, it was indeed respecting max flow rate. I'll provide reproducible steps shortly. It worked exactly as expected. Arachne/Classic definitely did not make a difference, it will change flow rates but doesn't stop the flow rate from exceeding the max flow in my testing. 

Try this, if you're still willing to investigate: 
With classic/arachne, set your flow ratio to 1.15 (just for demonstration). Then say you want to limit max flow to 25, set it to 25/1.125 = 2.222

You should notice the UI show flow rates now respecting that. I also noticed this on small outer perimeters -  where the line width was much higher than it should've been for a speed which would exceed the flow rate. It _does_ attempt to print at those flow rates, and the Preview shows the flow based on the gcode.

Give me a bit I wil check your project later, and also provide reproducible steps.

ShantanuNair, 2025-03-31:
@dewi-ny-je I went through your project, and could not replicate what you got wrt Arachne flows. With your 0.95 ratio, I got a gcode where the max flow is a bit above 14. And once I applied my solution, it did come exactly under 14mm3/s. Are you on the latest Orca? I'm on 2.3 which was just released, and am able to reproduce my solution consistently across projects.

FabioM3, 2025-03-31:
Now it makes sense, in prusa slicer there's the same problem, if you change the extrusion multiplier the max flow rate is not respected.

dewi-ny-je, 2025-03-31:
> [@dewi-ny-je](https://github.com/dewi-ny-je) I went through your project, and could not replicate what you got wrt Arachne flows. With your 0.95 ratio, I got a gcode where the max flow is a bit [...]

I opened the file I uploaded, sliced, and here it is, 20. 69 mm3/s max flow rate:

[image]

Maybe you mean actual flow rate used for printing, I mean max shown in the legend, used only for travel: I cannot see any ACTUAL segment with that speed, looking layer by layer. If I follow the print head, I see the limit of 14 mm3/s is broken only for travel:

At one instant it's printing at 13.30

[image]

the very next step is at 38 but it's not actually printing:

[image]

So yes, max actual print speed is limited by "flow limit x flow multiplier", while I would expect it to be "flow limit".

I think this is the bug you are describing above.

My bug is different, it relates to the gcode preview using (some, but not all!) non-print moves to define the legend for the flow graph.

ShantanuNair, 2025-03-31:
@dewi-ny-je Sorry, I don't understand how what I posted isn't relevant. OP says UI has been fixed previously but the issue is relating to line width x speed not matching line widths, which is what I discussed. 

[image]

Here's your project sliced in Orca. Which version of Orca are you using? Is that Bambu Studio? I don't think we can reproduce your display bug in Orca, as in the screenshot I have shown, it displays a max flow rate of 14.81.

Now the issue presented in the OP is present here, but minimally. To enforce an effective flow limit of 14, I set your max flow to 14*0.95=13.3 and it works to enforce it, addressing OPs issue.

[image]

To demonstrate, with your own project, the issue the OP is stating, in a clearer way, I increased your flow ratio and tested and was able to replicate the OP's issue as well as the fix. I think you are facing different issues as that is Bambu Studio.

SoftFever, 2025-03-31:
Hey guys,
The Max volumetric speed is actually ignored by design.
Once the test project is generated, you will notice that the filament profile's maximum volumetric speed is automatically set to 200 as below.
[image]

The test expected you to input the end volumetric speed in the dialog. You will notice some discrepancies between the end volumetric speed you input here and the max value in preview. This is because, in this volumetric test, we are using a very large layer height and line width, meaning that with an 1mm/s increase in speed, the flow will increase 1 or 2 depending on you nozzle diameter . However, the difference should not be too significant.
[image]

ShantanuNair, 2025-03-31:
@SoftFever Hey, appreciate all the work you do! 
I think the issue described is that max flow in preview pane crosses the max volumetric flow. I don't think there was confusion about whether the max-flow test should use the max flow, (ofc it should disable it for the test to run through). The confusion was in the misunderstanding that the volumetric flow limit, during regular printing outside of tests/calibration, is applied before the flow ratio is multiplied in. If a max flow of 25mm3/s is set, and I'm trying to push my printer to print just under the flow rate where my hotend would underextrude, I would expect gcode that doesn't cross that flow rate, however in actuality, it is the flow ratio multiplied by the max volumetric speed set that is the effective max flow the printer can be asked to extrude.

For example if I set max flow of 25mm3/s and flow ratio is 1.1, then the slicer will generate, at some places, line widths large enough to give a flow of 25*1.1 = 27.5. Now this is a problem if we're printing at tuned high flow rates, beyond which a hotend can't go.

dewi-ny-je, 2025-03-31:
> [@dewi-ny-je](https://github.com/dewi-ny-je) Sorry, I don't understand how what I posted isn't relevant. OP says UI has been fixed previously but the issue is relating to line width x speed not [...]

That's strange. I'm on 2.3.0-70931e5 (final).

I deleted the preferences completely, I added only a printer, then without snycing with the network plugin I loaded the 3MF I previously uploaded here (I actually downloaded it from here), opened it and pressed "Slice plate", nothing else.

I still get 20 mm3/s.

ShantanuNair, 2025-04-01:
@SoftFever Apologies for @-ing you again, but I think you previously replied thinking this was another issue/thread. What do you think about making this clearer to users? A lot of people set this setting after figuring out their hotend's max flow rate in an attempt to print at high speeds. Sometimes, they find a flow rate which gives consistent print quality across all features and set the max flow rate using max volumetric speed. But effectively, they may see higher flows, if their flow ratio is high. A lot of youtube videos have this explained incorrectly, none of them taking into account flow ratio.

On that note, I notice that regardless of max-flow rate, supports with adaptive layer height don't seem to take into the max flow rate into consideration at all and overshoot it significantly, even if the max is set to (desired_max_flow/flow_ratio), which should've accounted for surges above desired max flow.

dewi-ny-je, 2025-04-01:
Concerning the flow being "max flow times multiplier", since the multiplier is used to match expected with actual flow, the values displayed should not take the multiplier into account... the display of flow values multiplied by a calibration value is misleading.

ShantanuNair, 2025-04-01:
@dewi-ny-je It's getting a little tricky here - I see what you're saying, but wouldn't we want to show the real flow rate, and expect the max flow rate to limit the physical flow through the nozzle? The UI shows the actual flow rate. It's not the UI that's misleading, it's thinking that the max flow rate is respected (it is, but only in the context before the flow ratio is applied). It's the naming that's causing all the confusion.

SoftFever, 2025-04-01:
Regarding the original question of the post, in short, the equation is not as simple as multiplying the layer height by the line width and then multiplying by speed to equal the flow rate. No, the algorithm used in the slicer is much more complex. The equation you are using is just for rough estimation. If you want to learn more about the flow mass used in OrcaSlicer, as well as other slicers and derivative slicers, please refer to this wiki and the associated link posted in that wiki post.
https://github.com/SoftFever/OrcaSlicer/wiki/Precise-wall

P.S. When you are using Arachne engine, the flow math is even more complex. In short, the max flow rate won't be considered when there are very short line width changes. But that is fine because it's a very short period during which your hotend should have enough buffer to handle this.

dewi-ny-je, 2025-04-01:
> [@dewi-ny-je](https://github.com/dewi-ny-je) It's getting a little tricky here - I see what you're saying, but wouldn't we want to show the real flow rate, and expect the max flow rate to limit the [...]

The flow multiplier is supposed to be the "calibration parameter" which makes the actual flow identical to the expected flow.
It seems obvious to me that the gcode preview should show the expected one, not the "wrong one we provide the printer so that it can output the expected one".

dewi-ny-je, 2025-04-01:
> P.S. When you are using Arachne engine, the flow math is even more complex. In short, the max flow rate won't be considered when there are very short line width changes. But that is fine because it's a very short period during which your hotend should have enough buffer to handle this.

This might help explain why on Arachne I get 20 mm3/s even if I limit the filament to 14, while Classic brings much smaller discrepancies. They are still higher than the flow multiplier would make me expect, which means some short moves exceed it even with Classic. 

Still, this questions the usefulness of the gcode preview: if it uses some very short moves, exceeding the preset max flow by 60%, almost EVERYTHING else ends up being compressed into about half colour range, and I have troubles seeing the details I expect.

Maybe the gcode preview should set the max colour range using the 99th percentile, so that 1% of the moves end up compressed into the same red, which is no real issue after all.

FabioM3, 2025-04-01:
Prusa Slicer have in the preview: "Speed", "Actual Speed", "Volumetric Flow Rate" and "Actual Volumetric Flow Rate".

ShantanuNair, 2025-04-01:
**Also, demonstrating further, this isn't limited to Arachne, here's classic demonstrating failed max volumetric flows:**

[image]

Note, now when I half the Flow Ratio - the preview UI Scale is also halved. As well as limiting the highest flow effectively:

[image]

In both above examples, we use Classic and a max volumetric speed of 25. Contrasting both:
With 1.5 flow ratio, we can see the current segments with a flow of 37! And the range of the flow chart goes to 38.75.
With 0.75 flow ratio being half of 1.5, we see flow ratios of 18.75, half of 37! And the range of the flow chart foes to 19.37, again half. So it seems the max flow is indeed tracking proportionately, and it's unlikely the case that it's meant to give a flow of 38.75 when max is at 25. In classic mode. This is because the flow rate is so high (to demonstrate it tracking with flow rate). 

I still think there's something not factoring in flow ratio, and I think a fix would have widespread impacts for people trying to optimize printing at high flow but seeing underextrusion because it's crossing max flow. As per my understanding, either the flow ratio isn't being accounted somewhere along the max volumetric check pipeline, or the flow preview is showing us flow values incorrectly and incorporating flow ratio when it shouldn't.

I'd like to discuss this further if that's okay, and if there's a convenient way to converse, perhaps discord or something else do let me know.

Fwiw: Here are threads from Prusa and SuperSlicer discussing the same.
- https://github.com/prusa3d/PrusaSlicer/issues/10118
- https://github.com/supermerill/SuperSlicer/issues/3869

@igiannakas you [hinted](https://github.com/supermerill/SuperSlicer/issues/3869) that the gcode might already have embedded mm3/s in it, and that may be hinting this is a preview issue perhaps. If you have any insight here to offer, that would be helpful since you seem like you might be able to help piece things together.

stefansjs, 2025-04-01:
I think the issue here is simply a confusing UI. I think we should just use a consistent notion of flow rate throughout the UI, rather than having two different notions. If that's impractical for some reason then we should explain to the user why they differ. But it seems to me the gcode preview should show as accurate of information as it is capable, and the max flow rate should correspond to the *actual* max flow rate. Any argument for a different notion of flow rate should be explained to the user IMHO.

Regarding the rest of the discussion, I think we're currently adding a lot of noise to the thread. The problem IMO is not the precision/accuracy/correctness of the flow rate calculation, it is just the differences in user feedback. I simply want the two different boxes in the UI to use the same notion of flow rate.

dewi-ny-je, 2025-04-02:
But th preview should show the final, after multiplication, value, since it's the real one we expect from the printer, so the value without multiplication by the flow multiplier. It should not show some artificial number "we send the printer so the printer will do what we actually want".
We should show in the preview what we actually want...

ShantanuNair, 2025-04-03:
### I found the root issue, and a fix for while it's addressed. It is somewhat complex to explain the root cause, and in the heart of gcode.cpp.

Do you know there are two flow ratios? One is filament_flow_ratio and the other is print_flow_ratio. Filament ratio is what we set to increase flow after calibration tests. Print flow ratio is in process settings and we typically use it for object. 

- Filament Flow Ratio (Flow ratio under filament settings) is meant to be used such that when applied, the right amount of plastic is fed to the hotend when it would otherwise get too less/too much. So, the hotend gets the right amount of plastic. Speed capping occurs as if the standard, expected amount of plastic is fed in to the hotend.

- Print Flow Ratio, (under process settings) is meant to be used to tell the slicer you are sending in more plastic intentionally, and signals that (if >1), there will be more plastic that will actually be entering the hotend. In this case the speedcap is used considering that more plastic is actually flowing through the nozzle. 

Speed capping is done considering print_flow_ratio only and not filament_flow_ratio. 

https://github.com/SoftFever/OrcaSlicer/blob/70931e5321fa66966a5bfb251efca0e82307d427/src/libslic3r/GCode.cpp#L5318-L5324

Here, you can see it's divided by `_mm3_per_mm` which incorporates print_flow_ratio. 

So, if you want max_volumetric_speed to be enforced, you can use the per-object flow ratio, since that is print_flow_ratio, and bam, your speeds should be capped and extrusions should be calculated properly. @SoftFever What do you think about this recommendation? It seems that this was by design waay back in slic3r, and through the years the line between the 2 flow ratios/extrusion multipliers have become blurred.

Continuing [here](https://github.com/SoftFever/OrcaSlicer/issues/9184) for those that want to follow this issue relating to flow ratios, all eyes appreciated.

ShantanuNair, 2025-04-14:
Would appreciate if anyone can test https://github.com/SoftFever/OrcaSlicer/pull/9218 if works as expected/fixes your issues and improves print quality.

FabioM3, 2025-04-14:
Can you suggest some tets?

Smilefobia, 2025-04-18:
> Would appreciate if anyone can test [#9218](https://github.com/SoftFever/OrcaSlicer/pull/9218) if works as expected/fixes your issues and improves print quality.

It works the same as before your updates for me. 
(OrcaSlicer Nightly Windows Portable)

[image]

igiannakas, 2025-04-18:
Try this one here if you can @Smilefobia https://github.com/SoftFever/OrcaSlicer/pull/9217

closed 2025-07-27 by github-actions[bot]