OrcaSlicer/OrcaSlicer #7310: Add Separate Z speed and Z moves from XY moves!

Battou2501, 2024-10-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Orca Slicer combines XY move with Z move and sets max speed to it. For example this is layer change line from gcode^

G1 X160.468 Y156.291 Z9 F12000

F12000 for Z move??

I get very ugly walls, especially on corners. PrusaSlicer separates XY and Z moves and I get much cleaner results with same quality and speed settings in it, but Orca Slicer has more oprions and I would like to use it, but cant.

Here is a layer change from Prusa Gcode:

G1 Z13.256 F2400
G1 X166.781 Y150.892 F12000

 

### Which printers will be beneficial to this feature?

Klipper, Marlin, Others, All

### Describe the solution you'd like

Separate XY moves from Z moves and add a Z speed parameter like in Prusa Slicer

Rolerblader31, 2024-11-22:
I use Orca slicer and noticed that on layer changes the printer goes to the xy position first and THEN rises z. Looking at the gcode there are 2 G1 commands, first one x and y, second one just z. So they ARE different lines but in the wrong order in my opinion. Is there a reason for this order? I think the other way round should be better because the risk of knocking the print of with the nozzle is far less if the z movement is done first. (Image: look at third and second last line of code)
[image]

closed 2025-02-28 by github-actions[bot]

pyzimmer, 2026-09-14:
same problem please