OrcaSlicer/OrcaSlicer #7314: Slicer removes holes on the bottom of the model

voland696, 2024-10-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Generic Klipper printer

### How to reproduce

[image]
Loaded files with holes on the bottom surface...

### Actual results

[image]
 Holes on the bottom of the model disappeared after slicing. Both STL and STEP format affected

### Expected results

[image]
same model loaded to PrusaSlicer sliced properly.

### Project file & Debug log uploads

[Dual Portal Toy.zip](https://github.com/user-attachments/files/17581029/Dual.Portal.Toy.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

voland696, 2024-10-31:
[image]
same model sliced in PrusaSlicer

vgdh, 2024-10-31:
confirmed, something is broken.

buzzhuzz, 2024-10-31:
I've played with project file attached and found that this is "make overhangs printable" setting doing that funny thing. I'd suggest to disable it and use supports instead.

Diterex, 2024-11-04:
It also deletes added negative parts
[image]
[image]

closed 2025-02-10 by github-actions[bot]