OrcaSlicer/OrcaSlicer #7317: Crashing when slicing large projects

J3r4n, 2024-10-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 9 7900X 12-Core Processor
64GB RAM
Nvidia A2000

Tried the affinity settings suggested in Task Manager, tried changing power settings to Power Saving, tried using this C:\Windows\System32\cmd.exe /c start "OrcaSlicer" /Affinity FF "C:\Program Files\OrcaSlicer\orca-slicer.exe", and even wiped PC and running on a fresh Win11 install with latest BIOS and drivers. 

### Printer

Bambu X1C/Bambu A1

### How to reproduce

Project attached. I've had this issue with many large projects recently, seems to be worse with 2.2.0 and the latest Bambu Studio Beta. Slice project and wait.

### Actual results

Application will hang after a bit and eventually just close.

### Expected results

Slice slowly but complete successfully.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17587012/log.zip)
Project is too large to attach

### Checklist of files to include

- [X] Log file
- [X] Project file

J3r4n, 2024-10-31:
Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x7c 
Context :000000675AD78900 
ContextFlag : 0x10005f, EFlags: 0x10206 
Fault address:  0x159EFF46 0x1:0xEF46 C:\WINDOWS\SYSTEM32\ntdll.dll

Show CallStack:
7ffd159eff46:RtlProtectHeap [Unknown file][-1]
7ffd15a07afe:RtlAllocateHeap [Unknown file][-1]
7ffd15a07776:RtlAllocateHeap [Unknown file][-1]
7ffd13312391:GlobalAlloc [Unknown file][-1]
7ffcc8c6387e:DrvPresentBuffers [Unknown file][-1]
7ffcc897d3ef:Unknown [Unknown file][-1]
7ffcc897cfdb:Unknown [Unknown file][-1]
7ffcc8f3e09b:DrvPresentBuffers [Unknown file][-1]
7ffcc8f3ce07:DrvPresentBuffers [Unknown file][-1]
7ffcc8f3e80b:DrvPresentBuffers [Unknown file][-1]
7ffc4b7fd1b4:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc4b7f186b:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc4b6660c8:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc4b681fa4:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc4b67a0c1:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc4ca6e7a6:hid_write [Unknown file][-1]
7ffc4ca6374f:hid_write [Unknown file][-1]
7ffc4ca63b3b:hid_write [Unknown file][-1]
7ffc4ca63def:hid_write [Unknown file][-1]
7ffc4ca6367b:hid_write [Unknown file][-1]
7ffc4ca63aaa:hid_write [Unknown file][-1]
7ffc4c8593a5:hid_write [Unknown file][-1]
7ffc4c8593d0:hid_write [Unknown file][-1]
7ffc4c8593d0:hid_write [Unknown file][-1]
7ffc4c8593d0:hid_write [Unknown file][-1]
7ffc4c8593d0:hid_write [Unknown file][-1]
7ffc4c8593d0:hid_write [Unknown file][-1]
7ffc4c8b7119:hid_write [Unknown file][-1]
7ffc4caa8139:hid_write [Unknown file][-1]
7ffc4caa7f45:hid_write [Unknown file][-1]
7ffc4caa81ba:hid_write [Unknown file][-1]
7ffc4ca6f521:hid_write [Unknown file][-1]
7ffc4ca71e57:hid_write [Unknown file][-1]
7ffc4cacf4e8:hid_write [Unknown file][-1]
7ffc4b6c7bd8:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc4ad4ab12:Unknown [Unknown file][-1]
7ffc4adb8e62:orcaslicer_main [Unknown file][-1]
7ff6bff359d5:Unknown [Unknown file][-1]
7ff6bff35e96:Unknown [Unknown file][-1]
7ffd1506dbe7:BaseThreadInitThunk [Unknown file][-1]
7ffd15abfbec:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x159EFF46 0x1:0xEF46 C:\WINDOWS\SYSTEM32\ntdll.dll
0x15A07AFE 0x1:0x26AFE C:\WINDOWS\SYSTEM32\ntdll.dll
0x15A07776 0x1:0x26776 C:\WINDOWS\SYSTEM32\ntdll.dll
0x13312391 0x1:0x81391 C:\WINDOWS\System32\KERNELBASE.dll
0xC8C6387E 0x1:0xA3287E C:\WINDOWS\System32\DriverStore\FileRepository\nvwuwi.inf_amd64_bbc9fa39f6a68070\nvoglv64.dll
0xC897D3EF 0x1:0x74C3EF C:\WINDOWS\System32\DriverStore\FileRepository\nvwuwi.inf_amd64_bbc9fa39f6a68070\nvoglv64.dll
0xC897CFDB 0x1:0x74BFDB C:\WINDOWS\System32\DriverStore\FileRepository\nvwuwi.inf_amd64_bbc9fa39f6a68070\nvoglv64.dll
0xC8F3E09B 0x1:0xD0D09B C:\WINDOWS\System32\DriverStore\FileRepository\nvwuwi.inf_amd64_bbc9fa39f6a68070\nvoglv64.dll
0xC8F3CE07 0x1:0xD0BE07 C:\WINDOWS\System32\DriverStore\FileRepository\nvwuwi.inf_amd64_bbc9fa39f6a68070\nvoglv64.dll
0xC8F3E80B 0x1:0xD0D80B C:\WINDOWS\System32\DriverStore\FileRepository\nvwuwi.inf_amd64_bbc9fa39f6a68070\nvoglv64.dll
0x4B7FD1B4 0x1:0xE0C1B4 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4B7F186B 0x1:0xE0086B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4B6660C8 0x1:0xC750C8 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4B681FA4 0x1:0xC90FA4 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4B67A0C1 0x1:0xC890C1 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA6E7A6 0x1:0x207D7A6 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA6374F 0x1:0x207274F C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA63B3B 0x1:0x2072B3B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA63DEF 0x1:0x2072DEF C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA6367B 0x1:0x207267B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA63AAA 0x1:0x2072AAA C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8593A5 0x1:0x1E683A5 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8593D0 0x1:0x1E683D0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8593D0 0x1:0x1E683D0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8593D0 0x1:0x1E683D0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8593D0 0x1:0x1E683D0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8593D0 0x1:0x1E683D0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4C8B7119 0x1:0x1EC6119 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CAA8139 0x1:0x20B7139 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CAA7F45 0x1:0x20B6F45 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CAA81BA 0x1:0x20B71BA C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA6F521 0x1:0x207E521 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CA71E57 0x1:0x2080E57 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4CACF4E8 0x1:0x20DE4E8 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4B6C7BD8 0x1:0xCD6BD8 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4AD4AB12 0x1:0x359B12 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4ADB8E62 0x1:0x3C7E62 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xBFF359D5 0x1:0x49D5 C:\Program Files\OrcaSlicer\orca-slicer.exe
0xBFF35E96 0x1:0x4E96 C:\Program Files\OrcaSlicer\orca-slicer.exe
0x1506DBE7 0x1:0x2CBE7 C:\WINDOWS\System32\KERNEL32.DLL
0x15ABFBEC 0x1:0xDEBEC C:\WINDOWS\SYSTEM32\ntdll.dll

hotellonely, 2024-11-05:
Having similar issue on M3Max 64GB when slicing large objects. With this stl file scaled at 135% it crashes when slicing.
[2021 16 inch macbook.stl.zip](https://github.com/user-attachments/files/17625795/2021.16.inch.macbook.stl.zip)

hotellonely, 2024-11-05:
It looks like a bug related only to 64GB RAM configuration. Found some other people complaining the same issue with, yeah, 64GB RAM.

J3r4n, 2024-11-05:
Does going above 64 help or staying below 64?

hotellonely, 2024-11-05:
> Does going above 64 help or staying below 64?
> […](#)

Looks like both works. It's somehow only affecting 64GB systems.

J3r4n, 2024-11-05:
I will double the ram and report back. Thanks! I have another much less
powerful system that slices just fine and it has 32GB of ram. Hopefully
you're right!

J3r4n, 2024-11-05:
Alright, just swapped out the 64GB for a 96GB set. I'm having the same
issue. I'm at a loss. I have another system with lower specs that slices
the same files without issues. Both are fresh Win11 installs and the same
Orca version.

MaskedSquirrel, 2025-01-20:
Having a similar issue. I don't think my project is too large though. When I hit slice it gets to 15% then crashes with not notification or error message. I could upload my 3mf file if it would be useful.

- System 

  - Provider 

   [ Name]  Application Error 
   [ Guid]  {a0e9b465-b939-57d7-b27d-95d8e925ff57} 
 
   EventID 1000 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x8000000000000000 
 
  - TimeCreated 

   [ SystemTime]  2025-01-20T03:29:59.3260671Z 
 
   EventRecordID 148490 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  28512 
   [ ThreadID]  22020 
 
   Channel Application 
 
   Computer xxxxx
 
  - Security 

   [ UserID]  S-1-5-21-3554933374-xxxxxxxxxxxxxxxx
 

- EventData 

  AppName orca-slicer.exe 
  AppVersion 1.0.0.0 
  AppTimeStamp 671e6394 
  ModuleName OrcaSlicer.dll 
  ModuleVersion 0.0.0.0 
  ModuleTimeStamp 671e6388 
  ExceptionCode c0000005 
  FaultingOffset 0000000000b6c3a8 
  ProcessId 0x4bdc 
  ProcessCreationTime 0x1db6ae9471ba580 
  AppPath C:\Program Files\OrcaSlicer\orca-slicer.exe 
  ModulePath C:\Program Files\OrcaSlicer\OrcaSlicer.dll 
  IntegratorReportId 49a41231-aaa3-42cb-a0fa-861f39adc7c3 
  PackageFullName  
  PackageRelativeAppId

closed 2025-04-29 by github-actions[bot]