OrcaSlicer/OrcaSlicer #7318: Export Model *and* Supports as STL or 3mf

BearThreat, 2024-10-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

OrcaSlicer currently allows exporting gcode and 3mf files. What if we added a dedicated button to export the model and supports as a STL file?

This seemingly small feature addresses a significant and long-standing pain point for a unique slice of users. A quick Google search around variations of "export supports to stl" reveals years of forum inquiries and reddit posts requesting this functionality. The ability to export models and supports an STL would unlock several key benefits:

Support Customization: Users could modify or refine auto-generated supports in external 3D modeling software like Blender or MeshMixer.
Slicer Compatibility: OrcaSlicer's robust support generation could be leveraged even if a user is constrained to another slicer for their particular printer.
Workflow Efficiency: This eliminates the tedious and error-prone process of manually recreating supports in other software. Tree supports are awesome and should be available everywhere
Undisputed Standard: Everyone who doesn't want to hand build tree supports or hate other slicer's support implementation would download OrcaSlicer. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Proposed Implementation:
Add an "Export Model and Supports as STL" button to the export options that would:
Take the current model + generated supports
Combine them into a single mesh
Save as a standard STL file

### Describe alternatives you've considered

An alternative process would be to convert .gcode.3mf files (with the supports) to an stl or other 3d modeling file. As an aside, I've been unable to find functional online 3mf to stl tools.

B-Man1, 2024-11-04:
I also want this functionality

closed 2025-02-10 by github-actions[bot]

timnolte, 2025-04-17:
This is also something I was searching for. It's a shame that the bots just close feature requests like this.