OrcaSlicer/OrcaSlicer #7323: Default temps for TPU Temperature Tower are bad

QuantuMatrix, 2024-10-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS Sequoia 15.0 (24A335)

### Additional system information

M1 Macbook Pro 2021

### Printer

Creality CR10 Smart Pro

### How to reproduce

1. Go to Calibration
2. Click on Temperature Tower
3. Click on TPU

### Actual results

[image]

A TPU temperature tower is listed for testing by default at 280-320 (I think these values were accidentally copied from PA-CF)

### Expected results

A TPU temperature tower should default to something 200-230, not 280-320.
Otherwise, somebody is going to have a really bad calibration tower for TPU if they don't change these defaults.

### Project file & Debug log uploads

I don't have a log file at that location and my project is empty and non-exportable. 
Just open any instance of Orca 2.2.0 and you will see the issue when trying to add the TPU Temp Tower.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

Username-unavailable1, 2024-11-10:
+1, same issue on Windows 10 2.2.0 
[image]

Kuba6249, 2024-12-21:
[image]
[image]

mikebignell, 2024-12-28:
I've just found this when calibrating too, slightly annoying. I'm a software dev so I'll fix and raise a PR when I get a moment

closed 2025-01-02 by Noisyfox

YaImedgar, 2025-03-06:
Hi there! I'm currently experiencing this bug on my latest version 2.2.0 on macOS.

fyi @mikebignell @Noisyfox 

[image]

[image]

Noisyfox, 2025-03-07:
This should be fixed in 2.3