OrcaSlicer/OrcaSlicer #7326: Slicer creates an additional clearly perceptible and visible Z seam

Hamsterberg, 2024-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 Virgina

### Printer

Bambu Lab P1S

### How to reproduce

slice and print the file with "scarf joint seam (beta) is enabled

### Actual results

slice and print the file with "scarf joint seam (beta) is enabled, an additional clearly perceptible and visible Z-seam is printed

with version 2.1.1 the additional clearly perceptible and visible Z-seam is not present

### Expected results

No clearly perceptible and visible Z-seam at this position

### Project file & Debug log uploads

[test_v2-2-0.zip](https://github.com/user-attachments/files/17596821/test_v2-2-0.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

[image]

[image]

xantrk, 2024-11-01:
I think this has something to do with pressure equalizer. When it's off, flow change is abrupt, so it creates an additional "seam".

When it's one, speed is weird, which again creates another seam worse than scarf.

[image]
^Pressure eq off.

[image]
^Pressure eq on.

Scarf speed seems to override the equalized speed.

bdwilson, 2024-11-08:
Same here. Bambulab A1.  2.1.1 scarf was tuned after much trial and error.  Nothing changed on 2.2.0. All variables look the same in preview.  Flow was the only one that looked a bit off.  

[image]

[image]

zarbam, 2024-11-11:
I'm having the same effect on my macbook on 2.2 but fine on 2.1 (windows)

Sidiusz, 2024-11-28:
Yep, same here
[image]
(scarf seam on the right)

shyblower, 2024-11-28:
Today I found out that in 2.2.0 you need to activate "Retract on layerchange", "Wipe while retracting" and a wipe distance of about 0.8mm to make scarf seams look as they did in 2.1.1 where activating "Wipe while retracting" made it look worse.

Hamsterberg, 2024-11-30:
[image]

However, your suggested settings are already activated by default in my printer profile.

shyblower, 2024-11-30:
> However, your suggested settings are already activated by default in my printer profile.

I had it deactivated in the previous releases since it used to work much better without but now it seems to work only when it's activated. I'll post an example when I get to my printer.

shyblower, 2024-12-02:
This is what it looks like, now that I've reenabled "retract on layer change" and "wipe while retracting" with 1mm wipe distance. On 2.1.1 I had to disable wipe to achieve a comparable result.
https://github.com/user-attachments/assets/337b9d2c-9b89-4d7f-bf22-73533a393ac9

xantrk, 2024-12-16:
Could it be Z-Hop? https://github.com/SoftFever/OrcaSlicer/issues/7380

shyblower, 2024-12-25:
Now even "Wipe while retracting" stopped working for me. Don't know why.
Now I also get the this clearly visible seam every time, like all the others here.

shyblower, 2025-01-01:
Seems to be caused by the priming (unretract) before starting the scarf slope. The length is mathematically calculated perfectly as it is equal to the retraction length but it seems that during retraction a very small amount of already printed material from the end of the line gets also sucked back into the nozzle and thus is pushed out again at the beginning of the next line. This amount is so small that it does not matter too much when the next line is drawn with the same vertical thickness as the previous one (the configured layer height) but since the nozzle starts the next line very closely positioned over the previous layer when the scarf seam is applied,  this unintentionally carried and unretracted extra material produces the blob since it has to expand far more in horizontal direction due to the very small gap between nozzle and the layer below.

This is a hypothesis that I have put forward while testing different configurations and the only one that produced perfect scarf seam was outer/inner wall printing order, no infill and NO retraction during layer change. This way the nozzle moved without retraction from the end of the previous layers inner wall to the next layers outer wall scarf seam start.
Doing anything that needed "unretract" directly before starting the scarf, produced the blob.

shyblower, 2025-01-01:
I seem to be following the right path because configuring a negative "extra length on restart" value in the printer settings or via the setting overrides in the filament settings thus effectively reextruding less distance then was retracted improves the situation.
But when you set this value too low, air bubbles will be mixed into the beginning of the next line which will result in the same kind of issue you get yourself in if your nozzle is oozing molten filament during travel moves.

I just wonder why it seemed to work perfectly before v2.2.0?

I've tested all this using PCTG filament. It may of course behave differently when using eg. PLA.

shyblower, 2025-01-01:
What currently works best for me with scarf seam is, inner/outer wall printing order and "retract on layer change" disabled but you have to ensure that if the layer change happens inside the same object that the "travel distance threshold" is configured high enough to not force retraction.

begna112, 2025-01-02:
@shyblower what kind of values are you using for the negative extra length and travel distance threshold? I got a very interesting/bad result with -1 and 1. 
[image]

xantrk, 2025-01-02:
Prusaslicer creates good scarfs though with identical setings, so I think this is a bug

begna112, 2025-01-02:
So I've been able to confirm your results, @shyblower. 

Sunlu PLA+, X1C, .4 high flow nozzle, inner outer inner.

I've found that setting the negative extra length to half the line width size (.42mm/2 = .21mm) gives nearly perfect results on outer walls. However, with scarf seam enabled for inner walls, this leads to noticeable gaps. Notably, my pressure advance is also .2, so it's possible that's the value it corresponds to, not line width.

[image]

(Without retraction and deretraction, I'm getting stringing that I otherwise wouldn't, but that makes sense.) 

Disabling scarf seam gaps seems to mostly negate the inner scarf issue and lead to better quality. Almost imperceptible seam without flash enabled and barely able to feel it. 

[image]
(Left with Seam Gap 10%, right with Seam Gap 0%)
[image]
(Left with Seam Gap 10%, right with Seam Gap 0%)
[image]
(Left with Seam Gap 10%, right with Seam Gap 0%)
[image]
(Left with Seam Gap 10%, right with Seam Gap 0%)

Caveats: 
* inside vase concave shape (overhang on outside of wall) still has a seam, but it's not the straight blob one from before. 
* Overhangs are under extruding a bit. Maybe cause inner/outer/inner. If I remember correctly, that was already a caveat of scarf seams.

shyblower, 2025-01-02:
@begna112 you only have to deactivate "retract on layer change", the retraction length should of course not be touched.

begna112, 2025-01-02:
> @begna112 you only have to deactivate "retract on layer change", the retraction length should of course not be touched.

Yes, I know. But for non-scarf seams, retracting on layer change results in best quality and 0 stringing _for my filament and printer_. Just a caveat for my tuning that others might also have.

dodox1, 2025-01-06:
This is my test print with scarf seams activated.
I tried to modify the G-code (line 3581):
...
G1 X102.71 Y112.381 Z2.8 E.00008
G1 E-1.5 F3600
**;G1 Z3.2 F9600 ; dodo change - disabled**
G1 X103.507 Y116.973 Z3.2
G1 Z2.8
...

So, there is Z movement simultaneously with X and Y movement, and this scar almost completely disappeared.
I'm not saying it is a universal and simple solution, but it just shows what could be the possible reason for that scar.

[image]
[image]
[tesst_obloucek_PLA.gcode.gz](https://github.com/user-attachments/files/18324157/tesst_obloucek_PLA.gcode.gz)

shyblower, 2025-01-09:
@AngryMammoth I've already found the git commit that's responsible for the mess. This commit tries to fix the bug where the nozzle moves through already printed stuff when it's traveling to the starting point of a scarf seam, but obviously does it improperly.
I haven't had time yet, though to look deeper into it and fix it.

begna112, 2025-01-09:
@shyblower could you link it? Would be interesting to see.

shyblower, 2025-01-09:
@begna112 https://github.com/SoftFever/OrcaSlicer/commit/cf6d9c7

shyblower, 2025-01-15:
@dodox1  https://github.com/SoftFever/OrcaSlicer/commit/cf6d9c7 is the reason why this happens. This commit tries to fix the problem where the nozzle would move through already printed parts when descending to the scarf seam start position during a travel move but also causes exactly the problem you've described.

dodox1, 2025-01-15:
What I showed is not that the nozzle travels through the printed material, but instead travels upwards on the spot.

shyblower, 2025-01-15:
@dodox1 yes and exactly this is accidentally caused by the commit I've linked, but only when a sloped or spiral z-hop is done.

I'm currently trying to fixing it in my own OrcaSlicer fork and will post a pull request here if I'm successful.

shyblower, 2025-01-15:
@dodox1
 Although the referenced commit actually fulfills its task correctly for normal z-hops, namely to prevent the nozzle from being driven through objects that have already been printed, it also causes the problem you describe due to the intentionally separated z movement.

It doesn't seem to become easy for me, though, to combine x,y and z travel to the scarf seam starting point in a downwards sloped movement and also prevent the nozzle from moving through already printed stuff.

shyblower, 2025-01-15:
@dodox1 and there is still the problem, I've mentioned somewhere above, where obviously some molten filament from the previous line end gets accidentally sucked in during retraction which is spit out again at the beginning of a new line. This is not so obvious when a normal line is printed, because of the relatively large cross section compared to the start of a scarf seem line.

nubnubbud, 2025-01-17:
I have also been experiencing this problem. I don't think it's due to retraction, totally if at all. I was able to reduce it by increasing jerk settings, then increasing seam gap. This is obviously the wrong solution, but it seems more likely to be something like oozing mid-retraction, and the extra volume escaping during de-retraction. retraction, in my experience, doesn't "suck" as much as it simply relieves pressure.

I am confident this bug is why "outer first" is suddenly a preferred way to print like this, and what I have been forced to use, as anything else triggers the "fix" to reduce 

I have an old CR-10S, and I can confirm it's happening, even with a bowden tube, and I even made a new firmware to test pressure advance, to no avail- it doesn't seem to be that, either. Interestingly, if it was just a re/detraction issue, I imagine a bowden tube setup would be getting different results, but I'm seeing a very disctinct, uniform bulge at the beginning of the scarf, just like a P1S. 

until this is solved, is perhaps the best option to rollback to 2.1.1? the 2.1.1 version, however, has a number of incompatibilities with other features... **for now, scarf seam users might be best off using the 2.2 RC version.** the full release had the change in its changelog, and the biggest change otherwise I can tell is the new PA calibration pattern.

as for a correct way to move the nozzle down, it should begin tracing along the perimeter it's about to start printing, and then dip down along it to the intended scarf joint start without slowing. This would solve several issues;
1. speed/flow variation issues as the nozzle goes from still at the beginning of the scarf joint to its end
2. sudden angle changes as it moves from a different perimeter to the outer one
3. by tracing the end of the perimeter it is about to print, you can guarantee the nozzle is only moving along an area without filament laid down.

but I might be missing something here.

nubnubbud, 2025-01-19:
I just did a small battery of tests. Even with an entire millimeter of extra advance after a retract, it did nothing much. https://github.com/SoftFever/OrcaSlicer/commit/cf6d9c7 is clearly the issue, as version 2.2.0-RC produces perfect results on my 7 year old CR-10S with custom firmware. Pressure advance had no major effect on it, nor did any retraction settings or wipes, but going to version 2.2.0 release version causes the issue again. 

If you use scarf seams, use the release candidate for now as a workaround. I don't find any changes between the release candidate and full release compelling enough to justify broken scarf joint seams for most users.

xantrk, 2025-01-23:
@SoftFever

shyblower, 2025-01-23:
@nubnubbud I'm also quite sure it is this fix that introduced the difference in behavior between 2.2.0 and 2.1.1.
I also have an explanation for this:

When the nozzle leaves the end of a line for a travel move to the next line then a tiny bit of molten filament stays on the nozzle tip, even if it did a wipe movement before lifting off (I've observed that when the nozzle leaves an object for another wipe moves are often carried out in the opposite direction than the line was drawn, effectively gliding over still molten material). Retraction also might have sucked in a fraction of the still molten plastic underneath it, as I've already suggested somewhere above.

When z-hop is done before starting the scarf slope e.g. when you print outer/inner wall order or "retract on layer change" is off and z hop height is greater then zero and also with with the fix in https://github.com/SoftFever/OrcaSlicer/commit/cf6d9c7 in any case, the nozzle is moved to the scarf slope start position in a safe distance above the previous layer and then lowered down to the start z position which is with the default of 10 scarf steps one tenth of the layer height. This flattens out the tiny drop of filament and expands it horizontally creating a blob which consequentially leads to the scar like seam.

Without the above mentioned fix when your wall order is inner/outer then the nozzle gets dragged through the already printed layer because the the scarf slope starts, as described above, at a lower position than the layer height. During this the molten remnants from the previous line get completely wiped of the nozzle and no blob is produced.

Also the above mentioned fix for the nozzle being dragged through the layer is not properly implemented since it only works correctly for legacy z hops but not for sloped or spiraled ones.

nubnubbud, 2025-01-24:
@shyblower I disagree. I was printing tiny dense rollers, and watching. there is no room for anything under the nozzle. I don't use z-hop or anything like that, and wiping didn't affect it either. it seemed that the nozzle just stayed there, extruding for a moment, before starting the scarf seam.

I never mentioned z-hops for a reason, they were never a consideration in my testing, and my suggestion did not include or consider them. if your printer is properly calibrated and your profiles are well made, z-hops should do pretty much nothing, as they exist to fight oozing and stringing from machines that don't have accurate extrusion or can't accommodate material oozing.

My solution was only to make sure, that no matter what, the scarf seam began in the middle of a continuous movement along a perimeter that was guaranteed not to have any extrusion, therefore eliminating the need for advanced algorithms, z-hop mechanics, and changes to the travel code altogether. Keep It Simple Stupid.

shyblower, 2025-01-24:
@nubnubbud then it's the additional molten filament, that gets sucked in by retraction from the still liquid end of the line and is reproduced during unretraction. That's why I could reduce the issue with a tine negative extra prime length.

nubnubbud, 2025-01-24:
@shyblower ah, perhaps. I still find that suspect, but I use a bowden, so for me retraction is all about eliminating pressure in the bowden tube and eliminating oozing right at a travel. Maybe that's why you would use z-hop and I can't- for me, because mine can only reduce the residual bend inside the bowden tube, not z-hopping is how I eliminate stringing, but a faster, less spring-like reaction might allow yours to actually suck. (...in the good way.)
also of note, I turned my retraction and pressure advance settings against the seam, and there was zero space where I would get somethng reasonable. it went straight from positive seam, to at least half a millimeter of missing wall. I do pride myself on my calibration, so my machine shouldn't be sloppy enough to jump around the .3mm in extrusion that can happen in.

regardless of the mechanism (though I suspect it's something along the lines of "it begins extruding for the scarf joint too early because of pressure advance taking effect before the scarf joint sets extrusion to almost nothing" or "it should be de-retracting as the move starts, not before") we can agree that it's extruding when it shouldn't, and it's because of that change, but we're not sure why.

shyblower, 2025-01-24:
@nubnubbud and the reason why it worked so well for many people in 2.1.1 because of the issue of the nozzle being dragged through the previous layer in certain configurations which effectively cleaned the nozzle and might even have pulled out the remnants of the slightly sucked in material of.the previous line end.

That's why it works again when I remove the above mentioned fix for this which is implemented in 2.2.0, and is not working correctly anyway when used with sloped or spiralled z hop.

xantrk, 2025-01-24:
@shyblower Do you know if prusaslicer implementation is similar? That works perfecty

shyblower, 2025-01-24:
@xantrk if PrusSlicer uses the same Implementation it's probably still working because they haven't tried to fix or even still not recognized the nozzle getting dragged through the previous layer issue.

SoftFever, 2025-02-15:
@shyblower Nice work on the tracing and investigation!  Have you made any progress on the fix by any chance? 

Meanwhile, I pull the PR's author @fritzw in the discussion

fritzw, 2025-02-16:
Interesting, and good detective work. I also noticed this but I didn't think to go back and check if it could have something to do with my fix.

Just reverting the fix seems to be a good solution for small models, but for larger models with multiple separate perimeters (think e.g. a filament spool with cutouts in the side wall), the original behavior causes major collisions and even print failures, as the nozzle drags through the previously printed perimeters with quite some force. At least on my printer.

Maybe a proper solution would be to do the downward move slightly inside the model (on the second or third outermost perimeter) next to the spot where the scarf seam should start. This would wipe the tiny blob on the second perimeter instead of the outer one, hopefully reducing the effect. Maybe additionally offsetting the move-down point 2-3 mm "backwards" (i.e. tracing the perimeter it is about to print like @nubnubbud suggested) could also help by reducing the sharp corner at the start of the scarf compared to just moving the move-down point inwards orthogonally. I hope that explanation makes sense...

nubnubbud, 2025-02-17:
@fritzw Well, I had several ideas. I'm not sure about the exact toolpath the code makes, but I imagine it could be turned from one angled line, to two lines- a normal travel, and a diagonal path for the last few millimeters. 

but I keep reiterating- I really am unsure of where the extra material is coming from (and why I keep grilling Shyblower on being sure something is the case without experiments), and unlike the others, I'm not willing to guess exactly why more filament comes out by the time the nozzle has begun the ramp. it could be;

- Jerk settings cause it to slowly come to speed, while extrusion is immediate, meaning too much initial extrusion.
- the nozzle is oozing before it lands (despite retractions or other things. are they taking effect or not?).
- it's probably not linear advance- I tested it and there were no points where it fixed anything.
- oozing could happen internally, and unretraction could push it out.

and probably some others.

igiannakas, 2025-02-22:
@fritzw @shyblower would the wipe inside before external perimeter help here? It lowers z and unretracts 1 nozzle width inside the model, then moves to the outer perimeter and starts printing. 

This wipes the nozzle inside the model before moving externally.

igiannakas, 2025-02-22:
This option here: https://github.com/SoftFever/OrcaSlicer/pull/3287

fritzw, 2025-02-23:
> This option here: https://github.com/SoftFever/OrcaSlicer/pull/3287

Nice, that sounds like what I tried to describe. Do you know how this implementation handles features that consist only of two outer perimeters (i.e. one perimeter loop)?

igiannakas, 2025-02-23:
> > This option here: [#3287](https://github.com/SoftFever/OrcaSlicer/pull/3287)
> 
> [@igiannakas](https://github.com/igiannakas) Nice, that sounds like what I tried to describe. Do you know how this implementation handles features that consist only of two outer perimeters (i.e. one perimeter loop)?

I think if I recall correctly, it will not execute for a single wall perimeter - take a look here: https://github.com/SoftFever/OrcaSlicer/pull/3616

shyblower, 2025-02-23:
@igiannakas, @fritzw  Yes, sounds definitely like what both of us were suggesting. I'll have to try it.

shyblower, 2025-02-27:
@igiannakas, @fritzw, @SoftFever Unfortunately "Wipe before external loop" does not help, in fact it looks the same as if done with "correct" z-hjopping and I believe I know why. Take a look at the attached picture and you will see.

I tried this with my build where I've removed the z-hop fix so that the nozzle moves through the layer as in 2.1.1.
Both with retract on layer change and thus z-hopping in this case deactivated.

Upper: no wipe before external loop
Lower: wipe before external loop

Imho it works in the upper example because the nozzle moves directly in a slope down to the start of the scarf thus effectively wiping the nozzle and starting the scarf without having to move down the nozzle in a vertical move directly over the starting point, thus not squashing any filament remnants into a blob.

In the lower example, with "Wipe before external loop" activated, it does the backwards wipe move, collecting material from the still liquid end of the line, carrying it in a straight horizontal line above the scarf start, lowering the nozzle vertically and thus squashing the tiny bit of collected molten filament into a rather large blob, because of the very low height of the first scarf segment.

[image]

igiannakas, 2025-02-27:
What’s puzzling me is how large that blob is. What does the seam look like without scarf?

nubnubbud, 2025-03-03:
at this point I would be asking for a slow motion macro shot of it as it happens. it could be that the nozzle presses too far down and ends up remelting and bulging the wall, or any number of things. we need better observations.

Sidiusz, 2025-03-26:
bump, the problem still exist

GandalftheGay, 2025-04-01:
any fix/good workaround for this yet?
just did many tests because i could not find good settings without the the left start of the scarf seam bulging. Upping the "Scarf start height" made it even worse while making the middle of the seam nearly invisible (=

if i can somehow help to fix this problem with testing, just tell me

shyblower, 2025-04-01:
The only thing that works for me was reverting the fix (that had been implemented since 2.2.0) for preventing the nozzle to move through already printed parts of the current layer when moving to the start of the scarf slope and deactivating "retract on layer change" to force exactly this behaviour which effectively wipes the nozzle clean and thus prevents the blob at the start of the scarf which is the definite cause of that seam.

Since the nozzle moving slightly through the current layer has never caused any harm to my prints yet as long as I activate retraction for the top layer, I'll leave it that way for now until I or someone else comes up with a better solution 🤷🏼‍♂️

zarbam, 2025-04-01:
> any fix/good workaround for this yet? just did many tests because i could not find good settings without the the left start of the scarf seam bulging. Upping the "Scarf start height" made it even worse while making the middle of the seam nearly invisible (= i used PLA and ASA for all my tests btw.
> 
> if i can somehow help to fix this problem with testing, just tell me
> 
> also: would be nice to have an option to activate wipe before external loop for normal seams while having Conditional scarf joint enabled.

Not yet that I know of. I'm stuck on 2.1 because of the products I sell rely on this to work

bdwilson, 2025-04-02:
> Not yet that I know of. I'm stuck on 2.1 because of the products I sell rely on this to work

Same here. Stuck in time and missing the new features.

daithi-coyle, 2025-04-07:
Nothing to add really other than another data point.
Getting my best results (still with a seam but at least not a second seam/bulge) with "Scarf: OFF", "Walls Printing Order: Inner/Outer/Inner"

[image]

I have not tried Scarf: On with "Wipe Before External", "Wipe on Loops" and reducing "Wipe Speed" but based on the above comments that won't make much difference.

pm0u, 2025-05-01:
Is there a tldr; on this? i did read but its a bit hard for me to follow. Is v2.1.1 the answer? Was the offending code reverted for 2.2.0/2.3.0?

daithi-coyle, 2025-05-01:
> Is there a tldr; on this?
Still an issue 

> Is v2.1.1 the answer?
It's the easiest solution 

> Was the offending code reverted for 2.2.0/2.3.0?
Not so far

pm0u, 2025-05-01:
@daithi-coyle much appreciated, I'll give 2.1.1 a shot and see what I get.

igiannakas, 2025-05-01:
@SoftFever @Noisyfox what do you think here? Should the z change be reverted? Or disabled when scarf seams are on?

shyblower, 2025-05-01:
@igiannakas I just reverted the z-hop fix (that actually does not even work correctly with spiralling and sloped z-hop) in my 2.3.0 branch some weeks ago and haven't been experiencing any issue so far.

pm0u, 2025-05-01:
I'm actually a bit confused - i went to do the same as @shyblower but the commit was unable to revert when i attempted. It looks like it already was, ahead of 2.3.0?

https://github.com/SoftFever/OrcaSlicer/pull/8478

Is there some other added code that is also contributing to this issue?

[10 more replies]