OrcaSlicer/OrcaSlicer #7329: Please be sure that opening or creating a new project document doesn't close an edited project document without warning the user that saving is necessary to avoid lost work.

loshlee, 2024-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

Any

### Additional system information

This is related to the application schema, not the host system.

### Printer

Any

### How to reproduce

1. Open an existing project document or create a new one.
2. Make changes to the project document by, for example, adding notes in the Notes area of the`Other` section of the Process properties.
3. Create a new project document or open another existing project document from the file menu.

### Actual results

Editing of the project document that was originally open is lost when another project document is opened or created, or when the close button is clicked in the project document window. This happens because the existing project document is closed automatically, without any prompt indicating that data will be lost.

### Expected results

There should be a prompt to save any project document that has been edited since it was last saved before any action taken (such as opening another project document) that causes the project document to be closed in such a way that the changes can be lost. Changes should belong to the project document, not a preset that can be created to add non-specific project document changes.

### Project file & Debug log uploads

This is a schema, or conceptual fault, not a runtime bug.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

The scope of object editing should be reviewed and updated so that the application observes the simple Mac interface guideline of avoiding loss of work.

Noisyfox, 2024-11-01:
[image]
Make sure you didn't accidently checked this option.
You can reset this option in preference.

loshlee, 2024-11-01:
No, I don't think so. Anyway, I went to Settings (⌘,) and clicked the "Clear" button next to both "Clear my choice on unsaved presets" and "Clear my choice on unsaved projects", then quit and relaunched Orca Slicer (which launched a new project), typed a phrase into Notes area as described in the bug report, used "⌘Q" to quit, yet no warning appeared.

Noisyfox, 2024-11-02:
Because you pressed ⌘Q immediately after typing the Notes, make sure you clicked somewhere after typing so it detect changes correctly.

loshlee, 2024-11-02:
Never mind. If notes can't be associated with the project document rather than presets, I don't see what good they are.

closed 2024-11-02 by loshlee