OrcaSlicer/OrcaSlicer #7332: Skirt was generated under object

ChrAdlerNorderstedt, 2024-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

MacOS 15.1

### Printer

Neptune 4 Pro

### How to reproduce

1. Slice
2. Look at the Preview

[image]
[image]

### Actual results

Skirt was printed under the object

### Expected results

Skirt was created around the object

### Project file & Debug log uploads

[Schlauch-Teile.3mf.zip](https://github.com/user-attachments/files/17604003/Schlauch-Teile.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-11-01:
@ChrAdlerNorderstedt
Luckily this is no bug.
Your parts are simply not oriented parallel to the build plate.
Please use the attached file.
I corrected the orientation using the **`Lay on Face`** gizmo.
1. Click the object you want to correct.
2. Click the **`Lay on Face`** gizmo.
3. Click the face it should be placed on.

The viewing angle is from below the parts, just so you know.

[Screencast from 2024-11-01 23-35-04.webm](https://github.com/user-attachments/assets/a6e12214-98c8-44ba-b478-6c5cc405e6fb)

Please rename the file before use by removing the suffix _`.txt`_.
[Schlauch-Teile.3mf.txt](https://github.com/user-attachments/files/17604608/Schlauch-Teile.3mf.txt)

Please close this issue, since it is no bug. 😊

ChrAdlerNorderstedt, 2024-11-01:
Thanks for explanation, lesson learned!

closed 2024-11-01 by ChrAdlerNorderstedt