OrcaSlicer/OrcaSlicer #7334: Arachne resulting in layer skips for small geometry

sneakyjoeru, 2024-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

bambulab a1 mini

### How to reproduce

Slice file with arachne settings on 0.2mm nozzle. 

### Actual results

On some layers infill and perimeters are lost. 

### Expected results

Uninterrupted vertically sliced model.

### Project file & Debug log uploads

[dude-with-spear.zip](https://github.com/user-attachments/files/17604674/dude-with-spear.zip)
[log.zip](https://github.com/user-attachments/files/17604677/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Geometry must have relatively small area in slice, but can occur randomly like in this case:
[image]
[image]

This happens a lot in spike types of geometry, like this one:
[image]
[image]

This also is the case for 0.4mm nozzles, and I've learned to check my models most of the times, but this layer skip on head is above what's normal, so I want it fixed please.

Noisyfox, 2024-11-02:
Try disabling "Only one wall on top surfaces"

sneakyjoeru, 2024-11-02:
> Try disabling "Only one wall on top surfaces"

That did seem to do it for the head. It didn't help with the spikes:
[image]

sneakyjoeru, 2025-02-01:
Still waiting

sneakyjoeru, 2025-05-03:
pls stop closing >_<

Derdness, 2025-07-23:
Hi, for me helped increasing Resolution in Strength settings
Before (res 0.012 mm):
[image]
After (res 0.001 mm):
[image]