OrcaSlicer/OrcaSlicer #7336: Issue where Orca Slicer 2.2 (release) overrides Bambu Studio opening files from web - setting governing association with files are no longer accessible

sfarkhan, 2024-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

macOS

### OS Version

15.1

### Additional system information

MacBook Pro M1 16" 16GB 

### Printer

Creality CR-6 SE (Klipper) and Bambu X1C

### How to reproduce

I use Bambu Studio on Makerworld.com to open projects. Once I updated Orca to 2.2 release, Orca now overrides all 3MF files from my tested web browsers - Chrome/Firefox/Safari.

### Actual results

Orca overrides and it didn't in 2.2. Beta. I have no idea how to access the settings to change these in Orca as they don't appear. I have attached screen shots from a friend's Orca and mine
[image]
[image]

### Expected results

I should be able to access the settings and the existing defaults shouldn't be overwritten or at least a prompt to change would be nice.

### Project file & Debug log uploads

N/A this is a UI issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sfarkhan, 2024-11-02:
I tried selecting "Developer Mode" and the Associate Web Links don't show up

sfarkhan, 2024-11-06:
Update, another friend on Mac OS can replicate this issue.

miguesan01, 2024-11-16:
Same problem with MacBook Pro M4 and Sequoia.

SHIELD1011, 2024-11-24:
Same Problem! Please fix this!

sfarkhan, 2024-11-24:
Additional information: This is not a file association issue as I've gone that route and associated all mf3 files with Bambu to see if I can circumvent Orca and it does not work from browser. It works from opening in the Finder by clicking an mf3 but any browser and it defaults to Orca. Finally, I contacted Apple support to ask if this is a system setting and they said there is no way to change this in the system aside from what I've already tried above and the issue is with Orca Slicer.

dedgedesign, 2024-11-26:
Same issue here, very annoying

BoliverRob, 2024-11-28:
Same here, and I have tried averything

NotAmaan, 2024-11-29:
Seems OrcaSlicer sets itself to open bambustudio links by default. (Ref: https://github.com/SoftFever/OrcaSlicer/issues/3753)

Eitherways, I was able to change it by installing SwiftDefaultApps from here: https://github.com/Lord-Kamina/SwiftDefaultApps

After installing, go to System Settings > SwiftDefaultApps > URI Schemes > bambustudioopen and select the "Viewer" as Bambu Studio.app

[image]

miguesan01, 2024-11-29:
> Seems OrcaSlicer sets itself to open bambustudio links by default. (Ref: #3753)
> 
> Eitherways, I was able to change it by installing SwiftDefaultApps from here: https://github.com/Lord-Kamina/SwiftDefaultApps
> 
> After installing, go to System Settings > SwiftDefaultApps > URI Schemes > bambustudioopen and select the "Viewer" as Bambu Studio.app
> 
> <img alt="Screenshot 2024-11-29 at 5 16 51 PM" width="886" src="https://private-user-images.githubusercontent.com/7458155/391074804-a54979fa-6921-4d2f-a919-faa61c1423b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5MDU1ODAsIm5iZiI6MTczMjkwNTI4MCwicGF0aCI6Ii83NDU4MTU1LzM5MTA3NDgwNC1hNTQ5NzlmYS02OTIxLTRkMmYtYTkxOS1mYWE2MWMxNDIzYjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjlUMTgzNDQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2MyNTY2YmU2MzdhNjFlM2FmODE2NzQxNGI1MzNjZTUwODBlZDYxNTM4MzJiNWJiMzA2YzQyNGY0ZjQ3ZDNiMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.z8XXckzlQlNmuM1UowwGZ5rUGTDRrjkZLFC6_Qnx92E">

It works! Thank you!

mherren, 2024-12-14:
Confirmed that this is still an issue, though `SwiftDefaultApps` takes care of it.

bergjet, 2024-12-22:
Same Problem here.

JonLaliberte, 2025-01-10:
Yup

aizatto, 2025-01-21:
Ran into this problem, solved it with a [reddit comment](https://old.reddit.com/r/OrcaSlicer/comments/1ghi6k2/issue_where_orca_slicer_22_release_overrides/m3mkupu/) to use [duti](https://github.com/moretension/duti/).

copying comment here

> The only way I have found to fix this involves Terminal, HomeBrew and a utility called duti
> 
> If you have Homebrew already installed add duti:
> 
> ```
> brew install duti
> ```
>
> Then run this command:
> 
> ```
> duti  -s com.bambulab.bambu-studio bambustudioopen
> ```
>
> If you want to use Orcaslicer then run:
> 
> ```
> duti  -s com.softfever3d.orca-slicer bambustudioopen
> ```

psiberfunk, 2025-03-05:
this is still a problem, just ran into this today, and it's very irksome on macos

shapeepo, 2025-03-14:
I'm having the same problem on the 2.3.0 rc, MacBook pro m2 pro

SoftFever, 2025-03-16:
It's tricky because macOS doesn't allow applications to register protocol handlers dynamically. On Windows, I added options so that users set the option by themselves.
But it's impossible on Mac, it has been decided when compiling the application.

Some users want OrcaSlicer to handle MakerWorld links on Mac, while others prefer not to...

https://github.com/SoftFever/OrcaSlicer/issues/3753#issuecomment-2366742614

SoftFever, 2025-03-16:
It's deprecated.
Mac OS newer than 12.0 don't support this API anymore.

I might revert the change for 2.3.1, so that On Mac Orca won't handle that protocol.
For now please use that command line method as a workaround.

psiberfunk, 2025-03-16:
Hrmmm is there no way to change it in system preferences then ? That’s where it was apparently moved to 

> On Mar 16, 2025, at 8:30 AM, Adam Lesser ***@***.***> wrote:
> 
> LSSetDefaultHandlerForURLScheme

luminescence69, 2025-04-22:
Is there a similar solution or workaround for Windows?
I uninstalled Orca and tried everything but couldn't get the Makerworld links to open Bambu Studio

drkpxl, 2025-06-02:
Homebrew option worked for me. I gotta make a shortcut so I can just swap easily.

avxmw, 2025-09-04:
Same problem on Linux with 2.3.1-alpha (6ae89f7). Very annoying.

chielboogaard, 2025-09-30:
Yes same problem still here on m2 mac. Very annoying indeed