OrcaSlicer/OrcaSlicer #7339: Orca Slicer Crashes when Accessing User Home

CitrusIntellect, 2024-11-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Arch Linux 

### Additional system information

Nvidia 4060
Intel Core i7-13700F
32GB Ram
Hyprland - Wayland
SDDM

### Printer

Creality K1C

### How to reproduce

Go into anyway to open model, enter user home folder.

### Actual results

Application Crash

### Expected results

Should access the user home folder

### Project file & Debug log uploads

add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.449: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.449: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.449: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.450: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.450: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.450: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.451: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.452: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.453: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.453: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.453: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:52.453: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 251 lines cut ...]

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:54.967: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:54.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:54.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:54.971: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:54.972: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:11178): Gtk-CRITICAL **: 17:15:54.972: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
/usr/bin/orca-slicer: line 5: 11178 Segmentation fault      (core dumped) $0-bin "$@"

Executed with:
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json orca-slicer

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Compiled from the AUR

CitrusIntellect, 2024-11-02:
UPDATE:
I cant access my users home folder (person who executed) but If i create another folder like dummy under /home I can access it

KaratsuFr, 2024-11-04:
workaround with __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 (for now)

nickermire, 2024-12-09:
I have the same issue on PopOS 22.04
Intel® Core™ i9-14900K × 32
NVIDIA GeForce RTX 4080

KaratsuFr, 2024-12-09:
same here the performance is not as good as it is on windows
this is clearly a workaround more than a solution

any update on the flatpack version instead of appimage (too many issue and not as good as flatpack)?

closed 2025-03-17 by github-actions[bot]