OrcaSlicer/OrcaSlicer #7340: Cannot scale objects by entering number in textbox

m-anish, 2024-11-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04

### Printer

Ender 3 KE

### How to reproduce

1. Select object, select 'Scale' in 'Prepare' view. 
2. Scale modal dialog appears
3. Object can be scaled by dragging, however, not by entering number manually in the text entry
4. This functionality worked in 2.1.0-beta, but seems broken in 2.2.0

### Actual results

Cannot scale object as the text entry field in the scale dialog is not editable

### Expected results

Object should be scaled as per the number entered into the text box

### Project file & Debug log uploads

[stevenson-screen.zip](https://github.com/user-attachments/files/17605875/stevenson-screen.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ShaneDelmore, 2024-11-02:
Might not be related, but I have run into an issue where editing by number appeared to not work but I found the actual issue was that the value did not update unless I tabbed out of the text box.  If I closed the scaling dialog without tabbing out the edit did not take effect.  Could that be your issue?  Or is it just that you cannot actually type in the text box?

m-anish, 2024-11-03:
I don't know what happened, but because of the bug I reverted to my 2.1.0-beta appimage, but now that I test it again in 2.2.0, the issue isn't there. 

I'll close this bug, and reopen if I have something more conclusive.

closed 2024-11-03 by m-anish