OrcaSlicer/OrcaSlicer #7341: Disable M73 Doesnt Work

Otscho, 2024-11-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Artillery Sidewinder X1

### How to reproduce

If i set the hook in the printer option "Diasbale set remaining print time" it doesnt work, my printer shows me although the remaining print time. It is since update 2.2.0 in 2.1.0 it was fine

### Actual results

I tryied to turn on and off but it doesnt work

### Expected results

The printer shouldnt show the remaining print time

### Project file & Debug log uploads

[Deckel.zip](https://github.com/user-attachments/files/17606279/Deckel.zip)
[log.zip](https://github.com/user-attachments/files/17606280/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2024-11-02 by Otscho

geodic, 2024-11-17:
Hi @Otscho, did you figure out how to solve the problem?

fausby, 2024-11-20:
On Artilery Genius same problem. And this issue is duplicated: https://github.com/SoftFever/OrcaSlicer/issues/7269

Africatuin, 2024-11-29:
I have the same issue with the same printer, Artillery X1

grebur, 2024-12-15:
I have the same issue. I messed around with the setting under **Printer settings** > **Disable set remaining print time**. Here is the thing: When I check the setting to "Disable" I still get one M73 per layer in the G-code. When i **uncheck** the setting I get **double M73 G-codes**. Like this:

```
M73 P8 R132
M73 P8 R132
```

Cleary this must be a bug. Is there another setting that controls the second M73 or somewhere else in the code that is responsible for that one?

Schildkroet, 2025-02-12:
Please check/test https://github.com/SoftFever/OrcaSlicer/pull/8388